RavenSystem/esp-homekit-devices #1: WiFi SSID

ageorgios, 2018-02-03:
How do you configure WiFi SSID / password?

RavenSystem, 2018-02-03:
To configure Wifi, device enters in access point mode, so you can connect directly to it, and a web page will appear automatically to let you to configure your wifi network.

ageorgios, 2018-02-03:
Uploaded the firmware on a Sonoff TH16, connected to power, no SSID from it comming... What SSID name am I looking for?

RavenSystem, 2018-02-03:
SSID is "SonoffTH <end_of_mad_addr>". It's 2.4GHz Wifi.

What is your UART output?

ageorgios, 2018-02-03:
`/home/ageorgios/esptool/esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash -fs 8m -fm dout -ff 40m 0x0 /home/ageorgios/temp4/esp-homekit-devices/sdk/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /home/ageorgios/temp4/esp-homekit-devices/sdk/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/Sonoff_TH.bin`

RavenSystem, 2018-02-03:
Your UART output when you power-on your Sonoff, after flashing.

ageorgios, 2018-02-03:
I dont really know how to do this.

The difference is that I used external esptool v2.3, and the command directly instead of the 
`make -C devices/Sonoff_TH test`
because the esptool included in the sdk is v1.2 and does not work, stopping at this stage:
```
make: Entering directory '/home/ageorgios/temp4/esp-homekit-devices/devices/Sonoff_TH'
esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash -fs 8m -fm dout -ff 40m \
	0x0 /home/ageorgios/temp4/esp-homekit-devices/sdk/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /home/ageorgios/temp4/esp-homekit-devices/sdk/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/Sonoff_TH.bin 
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0320
Writing 4096 @ 0x0... 
```

RavenSystem, 2018-02-03:
Ok. Don't worry. Follow those steps, assuming that your Sonoff is connected to your FTDI:

1. Disconnect your FTDI from USB.
2. Connect it to USB.
3. In your terminal, run:
```shell
cd ~/temp4/esp-homekit-devices/devices/Sonoff_TH/
export ESPPORT=/dev/ttyUSB0
make  monitor
```

Now you will see the UART output.

ageorgios, 2018-02-03:
```
ageorgios@ubuntu:~/temp4/esp-homekit-devices/devices/Sonoff_TH$ make monitor
/home/ageorgios/temp4/esp-homekit-devices/sdk/esp-open-rtos//utils/filteroutput.py --port /dev/ttyUSB0 --baud 115200 --elf ./build/Sonoff_TH.out
Opening /dev/ttyUSB0 at 115200bps...
```
No output..

I think I will use the precompiled firmwares you just uploaded

RavenSystem, 2018-02-03:
It's strange. Maybe your FTDI has not enough power for Sonoff TH. If you have connected sensor, please, disconnect it.

ageorgios, 2018-02-03:
Success with your precompiled firmware using this:
```
/home/ageorgios/esptool/esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash -fs 8m -fm dout -ff 40m 0x0 /home/ageorgios/Desktop/Sonoff_TH_v0.1.0.bin
```

RavenSystem, 2018-02-03:
Can you setup your wifi now?

ageorgios, 2018-02-03:
Yes

I am adding the accessory in Home app

What is the HomeKit password?

ageorgios, 2018-02-03:
021-82-017 I guess?

RavenSystem, 2018-02-03:
I'm writing the Wiki just now :)

https://github.com/RavenSystem/esp-homekit-devices/wiki/General-user-manual

ageorgios, 2018-02-03:
Cannot pair it with this number :(

RavenSystem, 2018-02-03:
Sometime pairing fails, in that case, you can unpower your device, repower it, and start HomeKit setup again (Wifi config keeps configured).

ageorgios, 2018-02-03:
Success after repowering the sonoff!
Thanx for the great work! Keep it up! I will test it and report my experiences!

ageorgios, 2018-02-03:
Sonoff TH16 is recognized as Sonoff Thermostat
Would be great to have option to configure it on demand (like just a switch)

RavenSystem, 2018-02-03:
I'm going to write specific user manual for Sonoff TH, but some info:
- Auto mode has no effect because device has only one relay.
- Sensor is read every 30 seconds.
- If you have connected a heater, then you must use Heat mode, and if you have connected a cooler, then you must use Cool mode.
- If sensor fails, relay turn off automatically.
- You can select operating mode using the Sonoff button (1 LED flash: off; 2 LED flash: heat; 3 LED flash: cool).

RavenSystem, 2018-02-03:
If you want a normal switch, you can flash the Sonoff_Basic firmware.

closed 2018-02-03 by RavenSystem