RavenSystem/esp-homekit-devices #4: Support for Sonoff 4ch / 4ch Pro

omarkmsft, 2018-02-21:
Would it be possible to add support for Sonoff 4ch / 4ch Pro?  Would be happy to help test on my device.

RavenSystem, 2018-02-21:
Of course that is possible to suport Sonoff 4ch. The problem is that I don’t have one. My devices are Sonoff Basic, S20, S22 and TH16. I want to suport dual and 4ch too. Your help is welcome.

Can you install the Sonoff_Basic firmware and post here the results? I need an initial idea and I need what are the similar elements with Sonoff Basic switch. 

Many thanks.

Arduingo, 2018-02-22:
These are the GPIO Used in the 4CH

GPIO13 | Status LED (blue)
GPIO12 | Relay 1
GPIO05 | Relay 2
GPIO04 | Relay 3
GPIO15 | Relay 4
GPIO00 | Button 1 / Flash
GPIO09 | Button 2
GPIO10 | Button 3
GPIO14 | Button 4

The 4CH uses a ESP8285, the PRO uses a ESP8285 and an STM32, STM32 is used for some features related switching relay modes, but as far I have read and remember from the schematic relays and buttons can be controlled from the ESP too

RavenSystem, 2018-02-22:
According to info provides by @Arduingo , I have released the code of Sonoff 4ch. Feel free to test it and comment (Try it on 4ch and 4ch Pro too).

Many thanks.

omarkmsft, 2018-02-23:
Thanks RavenSystem I will try this weekend.  I have 4ch Pro.  Will let you know result.

Arduingo, 2018-02-23:
@omarkmsft maybe the PRO has a couple of relay signals inverted, please take not of that, also if you can try de behavior with different dip switch configurations will help adjust the code

omarkmsft, 2018-02-24:
@RavenSystem @Arduingo any chance one of you could build 4ch binary and drop in releases for me?  I'm trying to setup a build environment on my box but running into issues.  Would go faster if I could get a binary.

omarkmsft, 2018-02-24:
i was able to build fine so no need to drop a binary.  I'll try flashing sometime in next few days.

RavenSystem, 2018-02-24:
@omarkmsft Excellent!

omarkmsft, 2018-02-24:
Got it flashed to my Sonoff 4ch Pro today and was able to connect to Homekit directly from device - awesome!  Ch1/Switch 1 seems to be responding to DIP switch settings 2sec On then Off relay.  Apple homekit updates after 2 seconds of on which is cool.  

Somehow Ch2, Ch3, Ch4 - don't have that behavior.  Not sure if that is switch settings on board, or something to do with firmware.  All I use is Ch1 so ok for now.

omarkmsft, 2018-02-24:
btw this video is helpful in showing how to get 4ch Pro in boot loader mode for flashing - https://www.youtube.com/watch?v=hOFvbdYkOII

closed 2018-02-24 by omarkmsft

kiralikbeyin, 2018-06-28:
https://github.com/RavenSystem/esp-homekit-devices/issues/18

kiralikbeyin, 2018-06-28:
@omarkmsft Are you using v1.1 or v2 4ch-pro?

simoneorsi, 2019-01-16:
with the new ravencore firmware release all my sonoff 4ch pro stop to work.
Interlocking mode didn`t funcion more. I truy to downagrade the firmware but i connot do it!
any suggestion?
thaks

RavenSystem, 2019-01-16:
@simoneorsi What version was you using before, and what are you using now?

simoneorsi, 2019-01-16:
@ José A. Jiménez, actually I have installed a new release 0.5.13, the old one was 0.4.10.

RavenSystem, 2019-01-16:
@simoneorsi There is a lot of changes since 0.4.10. Please, update to last version 0.5.14 and provide logs and details of your bug. I suggest to open a new issue with all info about.

simoneorsi, 2019-01-16:
@José A. Jiménez you know I can downgrade to 0.4.10 because all my sonoff control all roll-up shutter and now are all locked.
I try to downgrade with old file that I had backupup but  the son off after I configure ota repositopry and ota binary file stay freeze. 
thanks a lot

simoneorsi, 2019-01-16:
@ José A. Jiménez Ok, now I have download a new version 0.5.1.15, but I have the same problem. The interlock of the channel don`t function.