RavenSystem/esp-homekit-devices #6: Sonoff Basic does not pair with HomeApp

osvaldoasn, 2018-02-26:
Hi,

Sonoff Basic not pairing (verifying) with HomeApp:

>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: mDNS Exponential Back-off announcement 3 with delay 9 seconds and TTL=4500
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: mDNS Exponential Back-off announcement 4 with delay 27 seconds and TTL=4500
>>> HomeKit: [Client 4] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 4C855C29-F378-4D1E-8E4B-86BD6882B0ED
>>> HomeKit: Configuring mDNS
>>> HomeKit: mDNS goodbye announcement 1: Name=SonoffB 962952 c#=1ff=0id=50:CE:E6:DC:FD:15md=Sonoff Basicpv=1.0sf=0ci=8s#=1 Port=5556 TTL=0
>>> HomeKit: [Client 4] Successfully paired
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: mDNS goodbye announcement 2: Name=SonoffB 962952 c#=1ff=0id=50:CE:E6:DC:FD:15md=Sonoff Basicpv=1.0sf=0ci=8s#=1 Port=5556 TTL=0
>>> HomeKit: mDNS first announcement: Name=SonoffB 962952 c#=1ff=0id=50:CE:E6:DC:FD:15md=Sonoff Basicpv=1.0sf=0ci=8s#=1 Port=5556 TTL=120
>>> HomeKit: mDNS Exponential Back-off announcement 1 with delay 1 seconds and TTL=4500
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: mDNS Exponential Back-off announcement 2 with delay 3 seconds and TTL=4500

[image]

RavenSystem, 2018-02-26:
Please, erase your flash and try again with latest commit.

osvaldoasn, 2018-02-26:
@RavenSystem It works now!
Thanks.

closed 2018-02-26 by osvaldoasn