RavenSystem/esp-homekit-devices #7: Sonoff Basic Wall Switch Problem

osvaldoasn, 2018-02-28:
Hi,

I'm testing a wall switch on GPIO14, but when i connect anything on GPIO14, without connecting GND Sonoff turn on and off continuosly, and when i connect GPIO14 to GND and try to turn switch on or off it goes on and off, same on HomeApp and Sonoff button.
When GPIO14 is disconnected it works ok but without the wall switch.

Thanks.

RavenSystem, 2018-02-28:
I don't understand you well. What kind of wall-switch are you using? You must use a normal on-off switch, not a button switch, and you must connect it to GPIO14 and GND, but not only to GPIO14 (I don't know how this is possible). I have tested just now, and works as expected. Each time the wall-switch is operated, Sonoff changes its status. Can you provide more detailed info, please?

osvaldoasn, 2018-02-28:
Hi, 
I'm sorry the problem was the wire that i was using.
Thank you.

osvaldoasn, 2018-03-05:
Hi,
I'm trying to use a ON/OFF wall-switch with GPIO14 and GND, now it works to turn ON and OFF, but when the wall-switch is in OFF position I can't turn ON with HomeApp, it turns ON and OFF (same time) and remains OFF. When wall-switch is in ON position HomeApp work as expected.

Thanks.

RavenSystem, 2018-03-05:
Hello, 
Are you using the last version (binary or built by yourself)?
Thanks.

osvaldoasn, 2018-03-05:
Hi,
I builded it by myself.
Thanks.

RavenSystem, 2018-03-07:
I'm sorry, but I'm unable to reproduce your issue. Please, can you compile it with `export HOMEKIT_DEBUG=1` and then send logs when your issue occurs? 

Many thanks.

osvaldoasn, 2018-03-12:
Hi,
HOMEK IT_DEBUG=1 generates a file or I must to save monitor output?
I'll send videos reproducing bug and I've tried on two Sonoffs.
Thanks.

RavenSystem, 2018-03-12:
It is better that you save monitor output.

Thanks.

osvaldoasn, 2018-03-12:
Hi,

I've updated last comment with output.
With 3.3v it works better, but with 220v, even induction changes switch status.

Thanks.

RavenSystem, 2018-03-12:
Please, can you provide some photos about how you wired the circuit?

Many thanks.

osvaldoasn, 2018-03-12:
[image]
[image]
[image]

RavenSystem, 2018-03-12:
Yeah, now I can reproduce your issue!! Maybe I will need sometime to fix it, but I have made a little change. Please, try the latest commit and comment.

RavenSystem, 2018-03-18:
As far I can read [here](https://groups.google.com/forum/#!topic/sonoffusers/5vmtT6mRxHE), problem can not be solved in code. When my Sonoff is connected to 220v, induction interferences cause a lot of issues; for example, if I turn on lights of the room (not Sonoff), Sonoff toggle too automatically. I'm using 3 mts of wire for the wall-switch. I will try to connect an external ceramic capacitor to GPIO14 and GND to avoid interferences and test if this solves the issue.

osvaldoasn, 2018-03-19:
Hi, Maybe it could be solved by software because with this firmware (https://github.com/arendst/Sonoff-Tasmota) GPIO14 works ok as switch (and other functions), without induction bug.

Thanks for your work!

pcsaito, 2018-05-03:
@osvaldoasn I think I solved this with software only, if you still have this issue please try my branch and tell me if it solves for you:
https://github.com/pcsaito/esp-homekit-demo/tree/LPFToggle

RavenSystem, 2018-08-06:
Solved in Sonoff RavenCore firmware.

closed 2018-08-06 by RavenSystem

peros550, 2018-08-06:
Hi, sorry for this out of topic question, if sonoff is connected in 220V mains, how would you monitor the output in case of an issue?

RavenSystem, 2018-08-08:
You must connect UART without connecting 3.3V wire. And Sonoff and computer must connected to same 230VAC circuit to minimize risk. But yes, there is a little risk. I personally do that, but I use a low cost PC, not my iMac.