RavenSystem/esp-homekit-devices #10: flash problem

Kristian8606, 2018-05-01:
Will anyone be so good to help me make a flash.
I need detailed instructions for the whole procedure.

Kristian8606, 2018-05-01:
I managed to make a flash, but now I have another problem
I do not see WIFI. What am wrong

Kristian8606, 2018-05-01:
I've done this problem too ...
Your project is very good, now the next question is how to upload the project of maxximulin with a thermostat.

RavenSystem, 2018-05-01:
I’m sorry, but I don’t know what you mean. If you want a thermostat in your Sonoff, you have the Sonoff TH firmware with these functions: https://github.com/RavenSystem/esp-homekit-devices/wiki/Sonoff-TH

Kristian8606, 2018-05-01:
I'm testing him now thanks. I now see 3 thermostat devices, a relay and a sensor. I would also like to build an LED strip

Kristian8606, 2018-05-01:
Unfortunately, I'm not quite a good programmer to help you, but I can test it.
Regarding the thermostat I do not understand why I have an extra relay in the application home.
Can it be used and if not how to remove it?

RavenSystem, 2018-05-01:
What firmware have you used?

Kristian8606, 2018-05-01:
Sonoff_TH_POW I followed the instructions

RavenSystem, 2018-05-01:
Sonoff_TH_wPOW -> wPOW is an experimental feature that combine a fake motion sensor and a switch to create automations to help with power outages (wPOW = with Power Outage Warning). For a Thermostat function only you must use Sonoff_TH (without wPOW suffix). You have pre-compiled firmwares here: https://github.com/RavenSystem/esp-homekit-devices/releases

Kristian8606, 2018-05-01:
I uploaded the TH version, thank you.

Kristian8606, 2018-05-01:
Works perfectly with the DHT22 sensor

RavenSystem, 2018-05-01:
Many thanks. I have included that info in the wiki: https://github.com/RavenSystem/esp-homekit-devices/wiki/Sonoff-TH

Kristian8606, 2018-05-01:
Does Sonoff S20 Firmware Work in Sonoff TOUCH?
i'm sorry for my bad english.

Kristian8606, 2018-05-01:
It is based on the ESP8285 chip

RavenSystem, 2018-05-01:
Sonoff S20 Button and Sonoff Basic could work, but I haven't a Sonoff Touch to test it. Don't worry about ESP8285, that chip is an ESP8266 with integrated flash.

Kristian8606, 2018-05-01:
I will test it as soon as possible and report.

Kristian8606, 2018-05-02:
After about 5 to 7 hours of work, the device was turned off and did not respond.
Do you know what may have caused this problem?

RavenSystem, 2018-05-02:
It's strange. I use 3 Sonoff TH in my house with heaters, and I don't have any problem. When you installed the last firmware, did you erase flash before?

Kristian8606, 2018-05-02:
Yes  before I upload a new firmware.
Strangely, it happens in the morning between 5 and 7 hours.

Kristian8606, 2018-05-02:
The same problem happened to me with the esp32-idf.
I'll try to investigate this case to figure out the possibility of this being caused by a power cut.

Kristian8606, 2018-05-02:
If so, can this be set to restore its last state.
That is, when I stop the power and plug it in again, the accessory switches on.

Kristian8606, 2018-05-03:
This morning the thermostat was off. The problem is not in the power supply. The test thermostat is esp8266 with DHT22 attached and heater simulation LED what can be the problem and why does it turn off in the morning between 5 and 7 hours?

RavenSystem, 2018-05-05:
I have some Sonoff TH working without any issue. Can you send logs from your Sonoff?

Kristian8606, 2018-05-05:
Tomorrow I'll upload logs.

Kristian8606, 2018-05-06:
I switched on the monitor all night.
That night did not turn off.
I already suspect that the problem is in the hardware itself or the power cables, I will do more tests and report.

 `Kristians-MacBook-Air:Sonoff_TH kristiandimitrov$ make monitor
/Volumes/case-sensitive/esp-homekit-devices/sdk/esp-open-rtos//utils/filteroutput.py --port /dev/tty.wchusbserial1420 --baud 115200 --elf ./build/Sonoff_TH.out
Opening /dev/tty.wchusbserial1420 at 115200bps...
>>> Sensor: temperature 25.5, humidity 34.4
>>> Sensor: temperature 25.4, humidity 34.2
>>> Sensor: temperature 25.3, humidity 34.2
>>> Sensor: temperature 25.2, humidity 34.8
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> Sensor: temperature 25.1, humidity 34.9
>>> HomeKit: [Client 4] Get Characteristics
>>> Sensor: temperature 25, humidity 34.8
>>> Sensor: temperature 24.8, humidity 34
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> Sensor: temperature 24.8, humidity 34
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> Sensor: temperature 24.7, humidity 35
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Get Characteristics
>>> Sensor: temperature 24.6, humidity 34.4
>>> Sensor: temperature 24.6, humidity 34.9
>>> Sensor: temperature 24.5, humidity 35.2
>>> Sensor: temperature 24.4, humidity 34.7
>>> Sensor: temperature 24.3, humidity 34.8
>>> Sensor: temperature 24.3, humidity 35.1
>>> Sensor: temperature 24.3, humidity 35.3
>>> Sensor: temperature 24.2, humidity 35
>>> HomeKit: [Client 4] Update Characteristics
[... 469 lines cut ...]
>>> Sensor: temperature 20.6, humidity 38.6
>>> Sensor: temperature 20.6, humidity 38.5
>>> Sensor: temperature 20.6, humidity 38.5
>>> Sensor: temperature 20.6, humidity 38.5
>>> Sensor: temperature 20.6, humidity 38.6
>>> Sensor: temperature 20.6, humidity 38.6
>>> Sensor: temperature 20.6, humidity 38.6
>>> Sensor: temperature 20.6, humidity 38.6
>>> Sensor: temperature 20.7, humidity 38.7
>>> Sensor: temperature 20.7, humidity 38.9
>>> Sensor: temperature 20.7, humidity 38.9
>>> Sensor: temperature 20.7, humidity 38.9
>>> Sensor: temperature 20.7, humidity 38.9
>>> Sensor: temperature 20.7, humidity 38.9
>>> Sensor: temperature 20.7, humidity 39
>>> Sensor: temperature 20.7, humidity 39
>>> Sensor: temperature 20.7, humidity 38.9
>>> Sensor: temperature 20.7, humidity 39
>>> Sensor: temperature 20.7, humidity 39
>>> Sensor: temperature 20.8, humidity 39.1
>>> Sensor: temperature 20.7, humidity 39
>>> Sensor: temperature 20.8, humidity 39.1
>>> Sensor: temperature 20.8, humidity 39.1
>>> Sensor: temperature 20.8, humidity 39.1
`

closed 2018-08-06 by RavenSystem