RavenSystem/esp-homekit-devices #11: Thingspeak integration

jubudu, 2018-05-01:
Hi,
first of all: congratulations and thank you for this very stable software package! Great work!

Anyway, there is something I personally miss:
I would like to log my homekit temperature and humidity values also to thingspeak. 
While I am able to code a simple thingspeak client for the esp8266, I did not get it to work inside the homekit-devices-environment.
Can you please give me some hints, how to code the thingspeak post  inside the Sonoff thermostat example? Thank you!

RavenSystem, 2018-05-02:
I don't know anything about Thingspeak. Can you put here your working Thingspeak code with a little explanation?

jubudu, 2018-05-02:
Hi, thank you for the fast response. 
Attached there is a link to a code snippet from the Arduino/ESP8266 IDE, that I used for a non homekit connected logger. I would like to implement this into your homekit connected sonoff thermostat i.e. for logging each status update. Please find source code I cloned from this git:
https://github.com/squix78/esp8266-dht-thingspeak-logger

RavenSystem, 2018-05-02:
I have seen your code, and I’m sorry, but that code uses Arduino SDK and this HomeKit project uses esp-open-rtos SDK. You need to find or code by yourself an implementation of Thingspeak for esp-open-rtos to make possible including it with this HomeKit project.

closed 2018-08-06 by RavenSystem

peros550, 2018-08-06:
Hi, things speak is a very nice service that can be used to keep historic data of your sensor's values.
Basically in order to submit data to thingspeak service all it takes is to format a string with sensors data and some API key (which is unique for each user/sensor) and submit to things speak in the form of a http get request. Therefore the proper question would be how to make a http get request in the esp-open-rtos SDK.