RavenSystem/esp-homekit-devices #12: F/w v0.4 stops responding after 5-10 minutes

i3laze, 2018-05-06:
HomeKit.app tells "No response" for my Sonoff Basic, although I ping the device fine.
10 sec reset helps.. for half an hour.

I'm currently on Windows and can't build from source.
Could you please publish a fresh binary with official esp-homekit lib?

i3laze, 2018-05-11:
Sorry for the delay. Just uploaded a debug firmware to Sonoff basic. That's some WebGUI debug, or I should listen Serial TTL output with some baud rate?

i3laze, 2018-05-11:
The device works great and instantly, so far! That debug binary maybe somehow more stable 🥇 
Could't get a debug from TTL.  You'd better add a Wiki page how-to provide debug logs.

i3laze, 2018-05-14:
Please, describe, how to collect debug info. That would be useful for others too.
I also suggest running some tiny web GUI, where these logs could be viewed and downloaded remotely.

i3laze, 2018-05-18:
By the way, [Wiki](https://github.com/RavenSystem/esp-homekit-devices/wiki/Flashing-a-binary-build) tells to flash 8m, but actual Sonoff flash is 1m, could it cause out of space crashes?
Still no debug instructions..(

I’m moving to Tasmota+HomeBridge, extra layer, but it’s rock stable.

RavenSystem, 2018-05-18:
From the beginning, Wiki tells you how to get UART output: https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions

Run another web server is not possible due to low RAM available.

Wiki tells to flash 8m... yes, 8m = 8megabits = 1MegaByte. Actual Sonoff flash is 8megabits.

i3laze, 2018-05-18:
Thanks. I've heard around, that TLS itself eats more resources than a web server.
I was trying to get a monitor output in Arduino IDE with no success.. will try to get some output from URL above.

RavenSystem, 2018-05-18:
You can monitor output with Arduino IDE. You must select the correct TTYPort and set speed to 115200.

reinisb, 2018-06-05:
Hello,

I am having the exact same issue as @i3laze. I flashed my v1.6 Sonoff Basic with v0.4 binary build via macOS terminal esptool.py. Everything works great after the setup for a couple of minutes, but then it simply stops responding and shows "No Response" at the Home app. Moreover, I also noticed that in addition to not responding the wifi "Sonoff-24E181" becomes available, but nothing happens when I connect to it. Finally, the 10 second reset helps, but after setup and working perfectly fine for 10 minutes or so, it crashes again.

I would be more than happy to provide debug information so that this could be fixed. I would truly appreciate if someone could please let me know how to collect debug information via macOS.

Thanks,
Reinis

[image]

reinisb, 2018-06-05:
Alright, so what I now flashed the Sonoff Basic once more with the Sonoff_Basic_debug.bin. The issue persists, however it is a bit different. Here are the steps:
1. Flash sonoff with Sonoff_Basic_debug.bin
2. Connect sonoff to 230V.
3. Connect to sonoff's wifi, the sonoff's GUI opens where I select the wifi and enter passoword.
4. Set up sonoff via the Home app on my iPhone.
5. Sonoff works - I can turn it on and off via the Home app.
6. I repeat turning it on and off ~5 times and it becomes unresponsive.
7. I unplug/plug back in sonoff to 230V - now the sonoff's wifi reappears and once I connect to it, I again see Sonoff's GUI where I can select the wifi and enter password. In other words, it seems that the sonoff simply crashed in step 5 and reset to setup mode.

I am ready to provide logs so this can be fixed, hence I would truly appreciate if someone could please provide step-by-step instructions about how to read the log from the sonoff (I suppose there is a log in there because it runs Sonoff_Basic_debug.bin).

Thanks!

RavenSystem, 2018-06-06:
Can you try the Sonoff S20 firmware and tell results? Don't worry about model; you can use it with Sonoff Basic.

reinisb, 2018-06-06:
@RavenSystem - I just now successfully flashed the Sonoff Basic with Sonoff_S20_Button_v0.4.bin. After connecting the Sonoff Basic to 230V the Sonoff Wifi appeared and I was able to select the wifi and enter password. Then, in the Home app on my iPhone I saw the Sonoff, entered the code and started waiting for setup to finish "Adding Sonoff 24E181". Unfortunately, it never did finish the setup and then simply showed "Couldn't add Sonoff 24E181". I pressed the physical Sonoff Basic for 10 seconds, but nothing is happening - the Sonoff Wifi does not show up. I guess I have bricked the Sonoff Basic. Perhaps re-flashing will help.

Suggestions are welcome. Thanks

[image]

RavenSystem, 2018-06-06:
@reinisb erase flash (`esptool.py -p <ESPPort> erase_flash`) and reflash it, please. Then, if HomeKit setup fails again, turn off and on and try again.

reinisb, 2018-06-06:
@RavenSystem I already erased flash before flashing with Sonoff_S20_Button_v0.4.bin.

reinisb, 2018-06-07:
@RavenSystem - I removed the old Sonoff Basic device from Home app, then erased flash from Sonoff Basic, and in the end re-flashed the Sonoff Basic with the Sonoff_S20_Button_v0.4.bin. Now everything is working perfectly fine - I managed to successfully select wifi/enter password as well as complete the setup at the Home app (this time the icon of the Sonoff Basic during setup was a round purple thing - I guess it is because of the Sonoff_S20_Button_v0.4.bin binary).

Bottom line - flashing Sonoff Basic v1.6 with Sonoff_S20_Button_v0.4.bin binary works perfectly fine. I will report back if issues arise. If I don't report back, then assume that there are no issues. Thanks!

reinisb, 2018-06-07:
Update: Alright, 10 minutse past and again there is "No Response" in the Home app. So the Sonoff_S20_Button_v0.4.bin did not help. Can someone else reproduce the issue?

What else can I try?

reinisb, 2018-06-07:
@RavenSystem In the [General user manual](https://github.com/RavenSystem/esp-homekit-devices/wiki/General-user-manual) it states "If configured wifi network is unreachable, device will enter in AP mode to setup wifi again." Could it be that the sonoff simply stops seeing my wifi network? I am using a brand new Netgear R7000P router and the sonoff is located 2 meters from the router - perhaps I need to change a setting in the router?

i3laze, 2018-06-07:
It could be anything, but I had the same symptoms.  I had no time to troubleshoot the issue, so I moved to Tasmota+HomeBridge.

Maybe you’ll  finish the bug hunt.
You have to disconnect mains power, connect TTL-cable and hope for some serial output (see terminal parameters some posts above)

reinisb, 2018-06-07:
@i3laze - I appreciate the input. Good to hear that it is not issue of the router. I am fully committed to finish the bug hunt.

@RavenSystem - I can erase flash, then reflash with Sonoff_Basic_**debug.**bin.zip, then reproduce the issue. The question is - how can I read the output to paste it here? I see that there is a command to read UART outpot at the [Build instructions](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions) page, but as I have not done any of the six steps and don't know the <device_name> I cannot perform the command (I am using stock Terminal on macOS High Sierra, hence I don't know if it makes sense to do those commands in the first place.).

Is it perhaps possible to read the UART outpot via esptool.py? They have some [basic comands](https://github.com/espressif/esptool#readme) as well as [advanced commands](https://github.com/espressif/esptool/wiki/Advanced-Options). Would any of these allow me to read the debug output? (Sorry, I know that I do not posses required technical knowledge to figure this out by myself).

reinisb, 2018-06-07:
Alright, I managed to read the output. I erased flash, then reflashed with Sonoff_Basic_DEBUG.bin.zip, completed the setup, and waited until Sonoff Basic woult not respond anymore and enter the AP mode (I took only a minute - I simply turned Sonoff Basic on and off via the Home app around five times until it stopped responding). Then, I unplugged Sonoff Basic from mains (230v) and without entering the flash mode I simply plugged the TTL adapter with sonoff attached to it into my mac's USB port and run **screen /dev/tty.usbserial-FT1BN31F 115200** in Terminal.

Here is the output from Terminal. I do not understand it hence I am looking forward to feedback!
[sonoff.log](https://github.com/RavenSystem/esp-homekit-devices/files/2081014/sonoff.log)

RavenSystem, 2018-06-07:
@reinisb good job, but the problem is that necessary logs must be taken when Sonoff fails, not then. Can you reproduce the issue when Sonoff is only connected to your TTL adapter?

reinisb, 2018-06-07:
@RavenSystem - oh alright, I did not know that it is actually possible to operate the Sonoff Basic while being powered by TTL adapter's 3.3V (aka without mains 230v) - somehow I had not thought about this.

Anyway, I erased flash and reflashed the Sonoff Basic once more with the debug binary and managed to get it set up and working while being powered by the TTL adapter. Now, whenever I press the on/off button via the Home app or physically on the Sonoff then I see real-time output in Terminal, which is pretty cool. The strange thing however is that while the Sonoff Basic is powered by TTL everything is working perfectly fine - I have pressed the on/off buttons via the Home app and physically hundreds of times over an hour period and everything is working perfectly. I cannot get it to crash (hence there is no crash info to report here from the output log)!

However, when I disconnect the TTL cable and connect the Sonoff Basic to mains power (230v) and then switch it on and off multiple times, then it crashes and enters the AP mode in a couple of minutes time. What is more, I repeated the same process with other Sonoff Basics that I have and the same happens with those as well - so my Sonoff Basic is not faulty.

To sum up - at this point in time it seems that the Sonoff Basic is crashing (aka entering AP mode) only when powered from mains 230v, but working perfectly fine when powered from TTL. Do you have any ideas why this might be happening? Can you reproduce this with your Sonoff Basic? Thanks!

RavenSystem, 2018-06-07:
@reinisb The issue is caused by an AC interference. It’s a common problem that happens with some people. Personally, I haven’t that issue, and actually I use 3 Sonoff TH and 1 Sonoff S20. I’m working on an universal Sonoff firmware and I hope that I can implement a low pass filter to avoid AC interferences. If you want, you can help me to test it, because your Sonoff only works for a few minutes, and it would be very fast to check the AC interf. I can not test it because I can not reproduce it.

reinisb, 2018-06-08:
@RavenSystem - oh, I see! I could definitely help you by testing it! Just let me know when a new build is available!

PS This means that the tasmota firmware includes the low pass filter to avoid interference, right? cc @i3laze

i3laze, 2018-06-08:
I heard AC filtering is just about false stats triggers...

Anyway, maximkulkin has that pulled recently:
https://github.com/maximkulkin/esp-homekit-demo/pull/76

RavenSystem, 2018-06-08:
@i3laze AC filtering is just about false triggers, but here devices will run out of work because a false trigger activates the reset settings.

Pulled mentioned is not a good approach because it uses a task that constantly reads GPIO states using a endless loop, meaning that a lot of resources are used, and this way is not recommended by eep-open-rtos nor espressif.

i3laze, 2018-06-10:
I would prefer autonomous _stable_ HomeKit build too over Tasmota+HomeBridge.
So, if issue reveals itself only on AC mains, we agreed it is lack of filtering, yeap?
What’s next?

RavenSystem, 2018-06-18:
@reinisb please, try this firmware the you can.
[Sonoff_Basic_0.4.1beta.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/2112254/Sonoff_Basic_0.4.1beta.bin.zip)

reinisb, 2018-06-18:
@RavenSystem - thanks! I just now successfully flashed my Sonoff basic with Sonoff_Basic_0.4.1beta.bin.zip and set it up via Home app. Now, I will test it and report back with results. Fingers crossed!

reinisb, 2018-06-20:
@RavenSystem after two days of thorough testing I can confirm that the Sonoff_Basic_0.4.1beta.bin has been stable and the main issue of it going into the AP mode have been resolved. Thanks for that!

That being said, I have experienced another issue ~30% of the times when I open the iOS Home app. I think that we have all experienced the situation when opening the iOS Home app will display some/all devices with status "No response". The "No response" status then either goes away by itself after a couple of seconds or when the respective device icon is pressed. Well, in the case of Sonoff Basic running Sonoff_Basic_0.4.1beta.bin the status "No response" will be shown for some loooong 20-40 seconds during which it is not possible to turn the Sonoff Basic on/off. See the attached video zip where I managed to capture it happening - the Sonoff Basic running Sonoff_Basic_0.4.1beta.bin is "Living Room Switch". Any ideas?

[Example.mov.zip](https://github.com/RavenSystem/esp-homekit-devices/files/2119216/Example.mov.zip)

RavenSystem, 2018-06-20:
@reinisb Mmm... Maybe the slow response of Sonoff is caused by the low pass filter task used in GPIO14. I don't know if you use GPIO14; this is used to connect an external wall mount switch.

0.4.1 beta has 2 changes:
- Reset to factory defaults is managed by a timer instead a tick count variable.
- GPIO14 is controlled by a low pass filter task instead interruptions.

This new firmware 0.4.2 beta disables GPIO14 and low pass filter task, so you can test it:
[Sonoff_Basic_0.4.2beta.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/2119386/Sonoff_Basic_0.4.2beta.bin.zip)

The idea is to check if low pass filter task is using a lot of CPU and casing your Sonoff responses slow.

reinisb, 2018-06-20:
@RavenSystem - alright, thanks. I just now succsffully flashed my Sonoff Basic with Sonoff_Basic_0.4.2beta.bin and set it up in iOS Home app. I will now test it and report back with results.

PS I do not use GPIO14, however I am planning to use is at some point in the future (in a year or so).

RavenSystem, 2018-06-20:
@reinisb well, the idea is to keep GPIO14 working finally.

One question, what is your _DTIM value_ of your Wifi AP settings?

RavenSystem, 2018-06-20:
@reinisb Ok. I asked only because in my case (talking about certified HomeKit devices, not Sonoff) I don't have the initial "not response"; when I open Home.app, all my certified devices responses immediately without any delay time, and this is because I set DTIM to 1 (repeat, in my case).

RavenSystem, 2018-06-20:
@reinisb Another firmware to test. After testing 0.4.2beta, here is 0.4.3beta, that enables GPIO14 but with an interval of 50 milisecs instead 10 of 0.4.1beta.
[Sonoff_Basic_0.4.3beta.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/2120340/Sonoff_Basic_0.4.3beta.bin.zip)

reinisb, 2018-06-20:
@RavenSystem About DTIM. To make matters more clear, 95% of the times my other homekit devices do not show the "No response". Thus I think that my router is behaving correctly. Actually, when thinking about it, I think that the other homekit devices will show the "No response" right after I have enabled WiFi on my iPhone, which is actually expected and normal. However, Sonoff behaves differently - it will show "No response" when WiFi has been on for a long time and other devices are working perfectly fine. Moreover, after testing Sonoff_Basic_0.4.2beta.bin I can confirm that the issue still persists - the Sonoff Basic frequently shows "No response" - sometimes for 10-40 seconds, sometimes for minutes, sometime will only become live only after turn WiFi on/off on my iPhone.

I will now flash with Sonoff_Basic_0.4.3beta.bin and report back with results.

i3laze, 2018-06-20:
In case you got Mac around, I would suggest using a free network sniffer [Charles](https://www.charlesproxy.com/), which is quite easy to set up to log timing and decode SSL dialog between iPhone/iPad and Sonoff.
You may see that Sonoff doesn’t respond instantly, for example.

RavenSystem, 2018-06-20:
@reinisb Ok, I thought that your delay with certified devices happened always. Well, with your info, I can confirm that AC interference is not reseting Sonoff and GPIO14 works with the low pass filter task. Next, when you test 0.4.3 beta, test it connected only to USB too, and try to reproduce the "no response" delay.

reinisb, 2018-06-22:
@RavenSystem - I have been testing Sonoff_Basic_0.4.3beta.bin while having the sonoff basic connected to my computer via USB and I cannot seem to reproduce the issue of it "Not responding". The issue happened once, but I cannot reproduce it. I will be away for the weekend, so I will continue testing on Monday to put here more systematic results.

phlexx, 2018-06-23:
@RavenSystem I may have the same issue. My sonoff drops out seconds after turning it on and also when pressing the iOS switch multiple times. Sometimes I get a steady green light (sometimes for hours) until I shut off the mains. I think it’s related to the ac-interferences. I have 3 units in use, all having problems sometimes but only one of them has issues on a daily basis. Do you have a theory or can I help out by doing some testing? Thx in advance.

RavenSystem, 2018-06-23:
Ok. The best way to avoid interferences is with a hardware filter like this:
[image]
A capacitor combined with internal pull-up resistor is a hardware low pass filter.

I'm working on a new firmware that can be used in several Sonoff models, with some customizable options and Internet OTA updates, to avoid disassembling for wired flashing.

If you want, you can flash Sonoff_Basic_0.4.3beta.bin and test it.

phlexx, 2018-06-24:
@RavenSystem - Thx for the infos.
I flashed 2 sonoffs today to have a benchmark. One unit behaves as usual.

The problem child does funny things:
- turning on the fan via ios works.
- turning it of via ios makes it "click" but the flip switches back up and the fan keeps spinning.
(how is that even possible?)
- tried it multiple times. it works eventually.
- pressing the hardware button to turn it on/off works without a hitch.

it didnt drop out the network yet. (yay!)
how can i "hardreset" it? how long do i have to press the button?

Sorry for all my questions and thanks for your time! :)

RavenSystem, 2018-06-24:
> turning it of via ios makes it "click" but the flip switches back up and the fan keeps spinning.
> (how is that even possible?)
Because ac-interference.

> how can i "hardreset" it? how long do i have to press the button?
You must press and hold built-in button until you see led turn on for a couple of seconds. Then, you must configure your wifi settings again and HomeKit. But hard-resetting it will not help. It's necessary to improve low pass filter. Soon I will release a new firmware for Sonoff (An initial version), with OTA updates and custom setup.

peros550, 2018-07-22:
Hi, I had the problem as described in the title of this issue. I used the latest beta (0.4.3) and now  it is connected stable for more that a week. What is the difference between firmware 0.4.3 and Ravencore 0.1.0 ? 
In firmware 0.4.3 I am experiencing a very random phenomenon. Sometimes, when I turn on the Sonoff basic, it turns on and immediately turns off within 0,5 sec.  It does not do it very often, just thought I should mention it in case anybody experienced the same. 

All the best!

RavenSystem, 2018-08-06:
Sonoff RavenCore has a lot of enhancements. Others firmwares are unsupported.
Check this: https://github.com/RavenSystem/esp-homekit-devices/wiki/Sonoff-RavenCore

closed 2018-08-06 by RavenSystem