RavenSystem/esp-homekit-devices #13: Power Outage/No wifi issue

wmwood, 2018-05-07:
I'm sure that this is an issue with the underlying framework.

I'm running 0.4 on a few Sonoff Basics and Sonoff S20s.

This past weekend I had a power outage and I was expecting the sonoffs to reconnect to wifi after the outage but to my surprise it seemed like they tried to connect and failed so they went into setup mode which required me to walk through the house and unplug/replug all of them in.

Is there any change we can make that when a device cannot connect it periodically tries to reconnect to the last known wifi instead of trying failing and going into setup mode?

Just a suggestion - I'm aware that this should probably be file under esp-homekit but it seems like we get faster feedback over here; lol.

Thanks!

RavenSystem, 2018-05-07:
It’s very interesting. I think that, due to the power outage, your wifi AP boots after Sonoff, cause Sonoff to go into wifi setup because your wifi was not ready.

Your suggestion can be a good fix to solve that. It’s true that this issue is not HomeKit related, but it’s esp-wifi-config related (used in this repo).

I suggest you that could be great opening this issue here: https://github.com/maximkulkin/esp-wifi-config

I will talk with Maxim later about this.

Arduingo, 2018-05-08:
It happen also to me, I got a log but in my case since the sonoff was connected to a USB port taking logs what went completly off was the router, after some tries sonoff crashed and stayed there, not in setup.

i3laze, 2018-06-08:
It's a pity current release cannot run HomeKit and WebGUI on the Sonoff at the same time.

The behavior after power outage bothers me too.
By the way, in what state the device will return after AC power loss? Last state?

_Tasmota has an option in WebGUI to control such behavior - it could be Last/On/Off/and even Opposite to Last._

closed 2018-08-06 by RavenSystem