RavenSystem/esp-homekit-devices #14: Feature request: Sonoff 4CH support

i3laze, 2018-05-11:
Any way you could support one as 4 Different HomeKit devices?

RavenSystem, 2018-05-11:
What do you mean exactly?

i3laze, 2018-05-11:
How will Sonoff 4CH/4CH Pro behave, when I flash your binary? Will it work? How it would be displayed in HomeKit?

RavenSystem, 2018-05-11:
There is no options to configure like Tasmota. It only will display 4 switches in Home App. It’s like 4 Sonoff Basic.

i3laze, 2018-05-11:
4 switches - that would be fine.
So, it should be working out of the box with your binaries?

omarkmsft, 2018-05-13:
I use it with 4ch pro, each switch works just fine. Shows up as 4 individual switches and all the dip switch settings work fine. That said, I've had to reboot every few weeks because it loses connection with homekit, I'm running 0.4 version. My understanding is Tamsota doesn't work directly with homekit, you need a raspberry pi inbetween. The nice thing about this firmware is nothing else needed.

i3laze, 2018-05-14:
@omarkmsft , please check this thread for debugging the reconnection issue:
https://github.com/RavenSystem/esp-homekit-devices/issues/12

closed 2018-05-14 by i3laze

omarkmsft, 2018-05-14:
Ironically I had a power outage this last weekend, and sonoff 4ch showed disconnected after that.  It required a reboot from me to reconnect it to wifi.  I didn't bother to think the power outage might have something to do with the disconnection.