RavenSystem/esp-homekit-devices #17: Ota in Sonoff Raven_Core

sn0wst0rm, 2018-06-25:
@RavenSystem
Hello, I was reading your code to better understand the functionalities of the homekit lib. I'm really interested in including an ota upgrade method, I've already understood the ota update mechanism in open-rtos, but what I am not understanding is what you are doing in this example.
I've tried to flash it to figure out, for example, the meaning of the modified characteristics you have made. In home app, it isn't shown any of them, either figures any button to reboot the device.
So my question is: after implementing that ota function, needed to switch between ROMs, how do you actually upload and flash it?

Thank you in advance!

kiralikbeyin, 2018-06-25:
I will try this https://github.com/SuperHouse/esp-open-rtos/tree/master/examples/http_client_ota

RavenSystem, 2018-06-29:
OTA firmware update system used by Sonoff_RavenCore is this: https://github.com/HomeACcessoryKid/life-cycle-manager

kiralikbeyin, 2018-06-29:
@RavenSystem please can you put some screenshots or video to wiki? 
https://github.com/HomeACcessoryKid/life-cycle-manager#how-to-use-it is not enough for a beginner like me.

kiralikbeyin, 2018-07-03:
Need tutorial

RavenSystem, 2018-07-05:
OTA is not ready yet. When it is ready, a tutorial will be published.

RavenSystem, 2018-08-06:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Sonoff-RavenCore

closed 2018-08-06 by RavenSystem

kiralikbeyin, 2018-08-06:
I checked "ota_repo" word and only found in devices/Sonoff_RavenCore/main.c line 703
```status = sysparam_get_string("ota_repo", &char_value);```

Where do you "sysparam_set_string" ? 

Nothing about ota found in https://github.com/RavenSystem/esp-homekit-devices/blob/master/libs/legacy_config/content/index.html

RavenSystem, 2018-08-06:
It's explain in the wiki:
> Sonoff RavenCore is an unified firmware for several Sonoff devices, with OTA updates and some user settings. It uses a OTA update system called Life-Cycle-Manager (LCM) from @HomeACessoryKid.

kiralikbeyin, 2018-08-06:
Thanks this is very useful. 

Where is repo of otaboot.bin image how did you make it?

RavenSystem, 2018-08-06:
All is explained in the wiki:
> Sonoff RavenCore is an unified firmware for several Sonoff devices, with OTA updates and some user settings. It uses a OTA update system called [Life-Cycle-Manager (LCM)](https://github.com/HomeACcessoryKid/life-cycle-manager) from @HomeACessoryKid.
https://github.com/HomeACcessoryKid/life-cycle-manager
And there you can read:
> Meanwhile, the development area is at https://github.com/HomeACcessoryKid/ota 
And there you can find https://github.com/HomeACcessoryKid/ota/releases

i3laze, 2018-12-22:
A question:

If I flash main.bin, I will have no OTA (for doing f/w updates via Eve.app) like forever?

RavenSystem, 2018-12-22:
@i3laze I don't know. That kind of questions is better that you ask in official OTA repo: 
https://github.com/HomeACcessoryKid/life-cycle-manager

HomeACcessoryKid, 2019-02-19:
No, the ota concept depends on loading the ota-main program in the second partition.

if you start with ota-boot, that loads the ota-main in there and oat-main then loads the user-main in the primary partition
If you load the user-main directly into the primary partition, then no ota can happen

hope this is clear,
HacK