RavenSystem/esp-homekit-devices #21: the problem of choice. Esp8266 or esp32

jukorp, 2018-07-09:
I will build a dual channel pwm dimmer -> trailing edge dimmer. Include local dim up and dim down buttons.
esp32 is much faster?

RavenSystem, 2018-07-09:
Esp32 is much faster than 8266. Higher clock frecuency and dual core. And many improvements and Bluetooth support too. And mDNS used in esp32 HomeKit framework is full compatible with Apple Bonjour (it passes official Bonjour Apple tests).

jukorp, 2018-07-10:
This ravencode support only esp8266?
If i need esp32 :)

RavenSystem, 2018-07-10:
Sonoff RavenCore is only for several Sonoff devices, and they use esp8266.

I want to play with esp32, but for now I haven't a protoboard with an esp32 chip.

jukorp, 2018-07-10:
Maybe this is good?
I have this kind of myself.

Aliexpress or ebay about 5€

https://m.ebay.com/itm/ESP-32-ESP32S-Development-Board-2-4GHz-Dual-Mode-WiFi-Bluetooth-Antenna-Module-M/152479854433?epid=17016407741&hash=item238081ef61:g:HcsAAOSwDVtaVJGO

jukorp, 2018-07-15:
does something happen in lwip development?
is this helpful?

https://github.com/espressif/esp-idf/issues/1512#issuecomment-405095172

RavenSystem, 2018-07-21:
Sorry, I haven't any experience coding for esp32 yet. And yes, I'm going to buy a esp32 development board from Aliexpress asap.

closed 2018-08-06 by RavenSystem