RavenSystem/esp-homekit-devices #22: HomeKit Garage Door support request for Sonoff SV

ATLHacker, 2018-07-15:
I am using a Sonoff SV (Safe Voltage / DC up to 24 volts) loaded with Sonoff_Basic to Open my garage door.  It is working well but you need to turn it on then off to simulate the momentary contact it requires.  I have set it up in Isolated mode, so the relay is isolated from supply power (garage door opener needs two contacts to be closed momentarily activate - open/stop/close).
Ref: 
https://www.itead.cc/wiki/Sonoff_SV and it is powered from the garage door opener's 5V.

Would it be possible to have a HomeKit Garage Door version of this code?  
I notice it is defined in esp-homekit-devices/external_libs/homekit/include/homekit/characteristics.h .
Ref: 
https://github.com/RavenSystem/esp-homekit-devices/blob/master/external_libs/homekit/include/homekit/characteristics.h  

#define HOMEKIT_SERVICE_GARAGE_DOOR_OPENER HOMEKIT_APPLE_UUID2("41")

The Sonoff SV has 3 additional IO pins available (GPIO 4, 5 and 14), so it should be possible to connect two of these (GPIO 4, 5) to magnetic reed switches to detect open and close state.

I am aware you may not have a Sonoff SV.  I am willing to perform any required testing.

Thanks for your work on this project.

RavenSystem, 2018-07-21:
@ATLHacker it's a great idea! Now Sonoff RavenCore supports these device types (You can configure device type using ElGato EVE App, but I must write Wiki section explaining this and how OTA updates work): 
 1. Switch Basic (for Sonoff Basic or TH without sensor)
 2. Switch Dual (for Sonoff Dual)
 3. Button + Socket (for Sonoff S20, TH without sensor or Basic)
 4. Switch 4ch (for Sonoff 4ch /pro)

And these others are in TO-DO list:
 5. Thermostat (for Sonoff TH or S22)
 6. Switch Basic + TH Sensor (for Sonoff TH or S22)
 7. Irrigation System (for Sonoff Basic or TH)

Device type 8 will be Garage door (for Sonoff SV). I don't have a SV, so I will need your help for testing and documentation works.

RavenSystem, 2018-08-06:
To keep in my mind:

CURRENT_DOOR_STATE:
0 | "Open. The door is fully open."
1 | "Closed. The door is fully closed."
2 | "Opening. The door is actively opening."
3 | "Closing. The door is actively closing."
4 | "Stopped. The door is not moving, and it is not fully open nor fully closed."

RavenSystem, 2018-08-16:
Now I have a Sonoff SV, but I have not a garage door, so I need some info about:
1. When door is opening and button is pushed, what happens?
2. When door is closing and button is pushed, what happens?

peros550, 2018-08-16:
Normally in both cases a button press should stop the door.

RavenSystem, 2018-08-16:
And with another press, door will do oposite movement, no?

peros550, 2018-08-17:
Exactly!

ATLHacker, 2018-09-04:
**Garage Door Reed Switches**
Looking at the ESP8266EX data sheet, GPIO 4 and GPIO 5 have only a single function so are probably the best option (no conflicts) for connecting the door open/close switches.

[https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf]((https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf)

The GPIO pins can be programmed for pull-up / pull-down (internal resistor pulls GPIO up/down when not connected / switch open) which avoids the need for external circuitry:
Pull-up
    Will read high unless connected / shorted to ground.
Pull-down
    Will read low unless connected / shorted to 3.3V. 

For testing, you could use jumpers or some switches to simulate this.  On the PCB the pin columns are marked S (signal / GPIO), V (voltage / 3.3V) and G (ground / 0V) and the rows are marked IO5, IO4 and IO14 underneath.  Strangely the pictures on the Itead web site are marked IO3, IO4 and IO14, but the table matches my PCB.

[https://www.itead.cc/wiki/Sonoff_SV](https://www.itead.cc/wiki/Sonoff_SVhttps://www.itead.cc/wiki/Sonoff_SV)

Pull up is probably the safest option (switch connects GPIO pin to ground) as the power line is not connected to the switch cabling.  Pull down would require the switch to connect the GPIO pin to 3.3V.  Shorting 3.3V line would kill the power to the board.  In the case of pull-up, shorting the GPIO to ground or 3.3V would cause no harm.

I have purchased some reed switches which are switches that are activated when close to a magnet.  

[https://www.banggood.com/5-Set-Recessed-Door-Window-Contacts-Reed-Security-Alarm-Switch-Screws-p-971447.html?rmmds=myorders&cur_warehouse=CN](
https://www.banggood.com/5-Set-Recessed-Door-Window-Contacts-Reed-Security-Alarm-Switch-Screws-p-971447.html?rmmds=myorders&cur_warehouse=CN)

Mine are normally open (N/O) but some are normally closed (N/C).  So, for my reed switches, one would be closed (causing GPIO pin to be pulled low) when the door is open and the other when the door is closed.  During opening/closing, both would be open (high).  This would be opposite for N/C switches.
There would need to be a way of specifying or determining whether N/O or N/C switches are used (i.e. change a variable before compile or during initial setup).

closed 2018-09-04 by ATLHacker

i3laze, 2018-10-05:
Wow, great job here!
(Haven’t been around since spring)
Sonoff SV is the right choice for the task, cause most garage doors i’ve met have internal circuit running on 24V and sometime also have 5V pins.

FYI for those, who can’t wait:
I’m currently using [modded](url
https://www.instructables.com/id/Sonoff-Garage-Door-Opener/) Basic (SV will do this job out of the box) with one reed sensor attached to GPIO14 to determine “Closed” status.
Homebridge+Mosquito MQTT Broker+https://github.com/iomax/homebridge-mqttgaragedoor

My gate has an NC reed, but a garage door has a NO reed. Such difference should be handled in WiKi (like “use NC sensors only”) or on “setup” stage (mentioned in the WiKi).

gpeter62, 2018-10-31:
Garage Door (device 8) is installed with Sonoff Basic + 1 switch (closed door). Everything works fantastic!
Thank you!!!