RavenSystem/esp-homekit-devices #23: Sonoff basic is very often "Unavailable" (not responding) in Home app

Marse1, 2018-07-19:
First of all, thank you very much for this software.

Could you please help me to identify a source of the issue with the randomly 'not responding' Sonoff basic?
My Sonoff basic v1.1 device is flashed with Sonoff RavenCore v0.1.0 and it is very often displayed in Home app as 'Unavailable'. At the same time it works fine in Home app on iPhone of my wife. In few minutes, after closing and opening Home app, it could become unavailable on both iPhones and then randomly become available on one or on both iPhones. Both, my iPhone 5s and iPhone SE of my wife are running on iOS 11.4.

I have also experienced this issue with another Sonoff basic device. And it seems that it does not matter whether Sonoff basis is powered from AC or from USB-UART 3.3V converter.

I have rebuild Sonoff RavenCore v0.1.0 with uncommented EXTRA_CFLAGS += -DHOMEKIT_DEBUG=1 and EXTRA_CFLAGS += -DqDebugLog and got the following output:

*******
(Client 7 is iPhone of my wife, which currently can control this Sonoff)
>>> homekit_client_process: [Client 7] Finished processing
>>> HomeKit: [Client 7] Closing client connection

(Then I unlock my iPhone)
- matched '_hap._tcp.local.' PTR
- responded with 138 bytes err 0

(here i have opened Home app)
- matched '_hap._tcp.local.' PTR
- responded with 138 bytes err 0

(Sonoff is displayed as "Unavailable" and I'm closing the app)
>>> HomeKit: [Client 3] Closing client connection

(here I have locked iPhone and in few minutes made another try and opened Home app. Sonoff status was changed to "Updating" and then again was changed to "Unavailable")
- matched '_hap._tcp.local.' PTR
- responded with 138 bytes err 0
- matched '_hap._tcp.local.' PTR
- responded with 138 bytes err 0
- matched 'Sonoff 71FEC9._hap._tcp.local.' TXT
- responded with 127 bytes err 0

(sometimes after few unsuccessful tried i got the following, when the Home app is closed)
!!! HomeKit: [Client 3] Error reading data from socket (code 113). Disconnecting
>>> HomeKit: [Client 3] Closing client connection

Do you have any idea what is the reason of such behavior?
Thanks in advance.

RavenSystem, 2018-08-06:
It sounds like a networks issue. What is your DTIM wifi setting  configured in your AP/Router? How many HomeKit devices do you have?

RavenSystem, 2018-08-06:
Please, check if Sonoff RavenCore 0.2.0 fixes your issue.

Marse1, 2018-08-06:
I am using Apple Airport Express and, unfortunately, it does not have DTIM settings. Also, I have no idea what is the default DTIM interval for this router. Google didn't help. I will try to find alternative router and check how it will work.

Currently, I have only one active HomeKit device in my network, which does not work.

Today, I have flashed two different Sonoff Basic devices with the new Sonoff RavenCore 0.2.0 and the  "Unavailable" issue still persists. In addition, the relay switch does not change its state physically, but changes its status in the Home app. The relay switch works only when I press the device button. I have tried to change all possible settings in Elgato EVE App, but it didn't help. Is there something I missed to configure? Previous RavenCore version did not have such problem.

Thanks in advance.

RavenSystem, 2018-08-06:
Version 0.2.1 fixes the relay bug.

RavenSystem, 2018-08-07:
Apple Airport Express should be fully compatible with HomeKit, so I don't think that is the problem. But I don't know where is the issue. If you can, try another router, at least to ensure that the router is not the problem.

If you can change DTIM value with the new router, set it to 1.

peros550, 2018-08-08:
@Marse1 are you still having this issue using latest version? 

Is there any chance that this behavior may be caused by other non certified homekit devices? I had noticed this random unresponsiveness on a different home were there were 3 sonoffs basic 1x flashed with Maxim's latest firmware and 2x with Raven's legacy version 0.4.3 beta (precompiled). All of them seemed to be unresponsive very often the moment I opened Home app and after a while they were  back again reporting their status. 

Interestingly,  now that I am away this home and have access as guest over Internet (not on the same LAN) I never see them unresponsive. 

 The router was TP-Link Archer C7.

RavenSystem, 2018-08-29:
@Marse1, please, try latest firmware (v0.3.0) and comment.

Marse1, 2018-08-29:
Sorry for the huge delay in answer.

I have just checked the latest 0.3.0 firmware and the 'not responding' issue in my case is not fixed. Before this test, I have switched off all the wi-fi devices (another 7 sonoffs flashed with Supla firmware, TV, laptop, etc.). The tested sonoff was very close to my Apple airport express router. On the first iPhone, sonoff is displayed as 'Unavailable' and on the second iPhone it works perfectly. After closing and opening Home app on both iPhones, sonoff works fine on the first iPhone and is displayed as 'Not responding' on the second iPhone. Very rarely sonoff is available on both iPhones at the same time.
If needed, I can build your firmware with the extended logging and share the log file, generated during the testing.

Unfortunately, I have not managed to get alternative wifi router with DTIM setting. Tomorrow, I will try to share this sonoff with the latest 0.3.0 firmware with my friend just for the testing. I will write the result.

RavenSystem, 2018-09-21:
@Marse1 FYI: I was using a TP-Link Archer D7 and HomeKit devices responded after 2-3 seconds, but now I change it for an ASUS ROG Rapture GT-AC5300 and now all HomeKit devices respond instantly. I have many wifi devices, and my Archer D7 can not manage all of them properly.