RavenSystem/esp-homekit-devices #25: thermostat

Kristian8606, 2018-08-04:
Should I change the value to 1 if I want the thermostat to turn on when I plug it into the mains?

homekit_characteristic_t target_state = HOMEKIT_CHARACTERISTIC_ (TARGET_HEATING_COOLING_STATE, 1, .callback = HOMEKIT_CHARACTERISTIC_CALLBACK (on_target));

RavenSystem, 2018-08-04:
Yes, set it to 1 to set heating state when Sonoff TH turns on.

Kristian8606, 2018-08-05:
I want to add a button to your thermostat to increase and decrease the target temperature, but I can not.
Would you give me an elementary example of a button?

RavenSystem, 2018-08-05:
```
#include <stdio.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <espressif/esp_common.h>

#define MY_BUTTON_GPIO    10   // Using GPIO 10 for this example

void my_button_callback(uint8_t gpio) {
    print("Button pressed at %i", gpio);
}

void user_init() {
    uart_set_baud(0, 115200);

    gpio_enable(MY_BUTTON_GPIO, GPIO_INPUT);    // Not necessary for GPIO 0
    gpio_set_pullup(MY_BUTTON_GPIO, true, true);

    // Here you can use GPIO_INTTYPE_EDGE_NEG, GPIO_INTTYPE_EDGE_POS or GPIO_INTTYPE_EDGE_ANY. Feel free to try and compare.
    gpio_set_interrupt(MY_BUTTON_GPIO, GPIO_INTTYPE_EDGE_NEG, my_button_callback);
}
```

Kristian8606, 2018-08-05:
I'm sorry about the stupid question, but I got this
   Task stack overflow (high water mark = 0 name = "IDLE")

RavenSystem, 2018-08-05:
Can you put here your complete code? Or is it published anywhere?

Kristian8606, 2018-08-05:
`/*
 * Sonoff TH
 * 
 * v0.4
 * 
 * Copyright 2018 José A. Jiménez (@RavenSystem)
 *  
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 * 
 * http://www.apache.org/licenses/LICENSE-2.0
 
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

#include <stdio.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_common.h>

#include <etstimer.h>
#include <esplibs/libmain.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <wifi_config.h>
#include <led_codes.h>

#include <dht/dht.h>

#define BUTTON_GPIO         0
#define LED_GPIO            2//13
#define RELAY_GPIO          12
#define SENSOR_GPIO         14
#define ON_GPIO          4

#define DEBOUNCE_TIME       500     / portTICK_PERIOD_MS
#define RESET_TIME          10000   / portTICK_PERIOD_MS

#define POLL_PERIOD         2000

uint32_t last_button_event_time, last_reset_event_time;
float old_humidity_value = 0.0, old_temperature_value = 0.0;
static ETSTimer thermostat_timer;

void relay_write(bool on) {
    gpio_write(RELAY_GPIO, on ? 1 : 0);
}

void led_write(bool on) {
    gpio_write(LED_GPIO, on ? 0 : 1);
}

void update_state();

void on_update(homekit_characteristic_t *ch, homekit_value_t value, void *context) {
    update_state();
}

void on_target(homekit_characteristic_t *ch, homekit_value_t value, void *context);

void reset_task() {
[... 240 lines cut ...]
    
    name.value = HOMEKIT_STRING(name_value);
    serial.value = HOMEKIT_STRING(name_value);
}

void on_wifi_ready() {
    led_code(LED_GPIO, WIFI_CONNECTED);
    
    create_accessory_name();
        
    homekit_server_init(&config);
}

void user_init(void) {
    uart_set_baud(0, 115200);
    
      gpio_enable(ON_GPIO, GPIO_INPUT);    // Not necessary for GPIO 0
      gpio_set_pullup(ON_GPIO, true, true);
       gpio_set_interrupt(ON_GPIO, GPIO_INTTYPE_EDGE_ANY, my_button_callback);
    
    wifi_config_init("Sonoff", NULL, on_wifi_ready);
    
    thermostat_init();
}
`

RavenSystem, 2018-08-05:
Can you paste here your complete log?

Kristian8606, 2018-08-05:
`>>> Sensor: temperature 25.8, humidity 43.5
>>> Sensor: temperature 25.7, humidity 43.8
>>> Sensor: temperature 25.8, humidity 44.1
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Task stack overflow (high water mark=0 name="IDLE")
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Button pressed at 4>>> Sensor: targ 22, targ 22
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")
Task stack overflow (high water mark=0 name="IDLE")`

RavenSystem, 2018-08-05:
Try to change:
`gpio_set_interrupt(ON_GPIO, GPIO_INTTYPE_EDGE_ANY, my_button_callback);`
with:
`gpio_set_interrupt(ON_GPIO, GPIO_INTTYPE_EDGE_NEG, my_button_callback);`

RavenSystem, 2018-08-05:
And increase POLL_PERIOD to at least 5 secs.

Kristian8606, 2018-08-05:
I'd get clean code, it was a test and a mess

thanks

Kristian8606, 2018-08-05:
https://github.com/Kristian8606/Sonoff_TH/tree/master/Sonoff_TH

RavenSystem, 2018-08-06:
What GPIOs are used to control temperature? GPIO 4 and...?

RavenSystem, 2018-08-06:
I have made a PR in your repo.

closed 2018-08-06 by RavenSystem

Kristian8606, 2018-08-07:
You are a good man that you have done a good job for me thank you!