RavenSystem/esp-homekit-devices #26: Version 0.2.1 not very reliable for me

peros550, 2018-08-07:
First of all, many thanks on providing a firmware with so many features. 

If possible, please check out this video where I demonstrate a very strange behavior. 

https://www.youtube.com/watch?v=_JhKPpph3XA

The sonoff's output is connected to a power relay with a light indicator at the right top corner. 
You 'll see that no matter how many times I tried to toggle output, it didn't want to change, at least most of the times. 

When I tried the hardware button on the sonoff, changing state was successful with no issues.   

Any ideas?

RavenSystem, 2018-08-07:
I thought this issue, introduced in 0.2.0, was fixed in 0.2.1. Let me check it. Are you sure that you are in 0.2.1?

peros550, 2018-08-07:
Yes, I just installed OTA and double checked the version information in home app.  
once again many thanks for your efforts.

RavenSystem, 2018-08-07:
Please, can you tell me what device type are you using? And what settings are you using too (GPIO14, etc)?

peros550, 2018-08-07:
I am using Sonoff basic, and GPIO14 toggle setting is by default ON didn't change it. Everything is default as set after the OTA installation.

RavenSystem, 2018-08-07:
Please, can you disable GPIO14 if you don't use it?

peros550, 2018-08-07:
I disabled it but more or less the same issue. 
I need to mention that while I had GPIO14 setting to off, the hardware button on Sonoff was able to toggle output. Is this expected?

peros550, 2018-08-07:
Nothing special. The same setup (from an electrical connections perspective) have worked flawlessly with previous stock firmware. 

Give me some time to check : 

a) Electrical connections once again (now it's late here in Greece and can't do it. Will check in a few hours tomorrow morning)
b) Flash another sonoff and try that one. Truth is I didn't flash the one that was working, I preferred to flash a second unit. Maybe there is a hardware issue with that unit. 

Thanks for your time, I will let you know tomorrow.

RavenSystem, 2018-08-07:
I have been able to reproduce the issue, for a few times. The problem is that relay activation generates an interference that activates the button GPIO. I will try to fix it.

peros550, 2018-08-07:
Maybe it would be a good idea to temporary deactivate the GPIO14 when you set relay output trough the homekit callback and then have it activated (if it's all possible).

RavenSystem, 2018-08-07:
No, GPIO14 no, physical button of Sonoff is on GPIO0.

peros550, 2018-08-07:
Then I am terribly sorry. I was under the impression that hardware button was connected to the GPIO14. 
Really sorry for that.

RavenSystem, 2018-08-07:
GPIO14 is for an additional and optional switch.

RavenSystem, 2018-08-07:
I'm testing a fix.

RavenSystem, 2018-08-07:
It seems to work well. I'm going to release v0.2.2 just now.

peros550, 2018-08-07:
Great, update takes place though the Eve app? All it takes is to toggle the update switch and wait?

RavenSystem, 2018-08-07:
Yes, only set it to ON and wait for 5-6 minutes.

RavenSystem, 2018-08-07:
And please, let me know if this new version works.

peros550, 2018-08-07:
Sure, just waiting for the OTA to finish :)

RavenSystem, 2018-08-07:
Remember that you must close and open Elgato or Home App to reload the accessory after OTA.

peros550, 2018-08-07:
Hi again, 

I have tried many times ti initiate OTA. I can see from the router app that Sonoff downloads ~400 KB of data then wait then nothing. Restarted everything. I'll keep trying a few more times and let you know.

RavenSystem, 2018-08-07:
It's strange. I've updated my Sonoff S20 without issues. Try these steps:
1. From ElGato App, turn on Firmware Update.
2. Close ElGato App, wait about 10 seconds, and open Home App.
3. Wait until accessory becomes responsive.
4. Check firmware version.

peros550, 2018-08-07:
Finished update :) 
Firsts tests are positive. No issues. I 'll keep testing and let you know. 

Many thanks for your support. You are great.

RavenSystem, 2018-08-07:
More tests tomorrow. Thanks to you for testing and feedback.

peros550, 2018-08-09:
Hi Jose, 

I just wanted to let you know and because I think this is not an issue under your responsibility I 've chosen not to open a new issue but write this message in this closed topic. 

I believe something does not work well with the OTA mechanism. Two times I needed to update (0.2.1 >> 0.2.2 and 0.2.2 >>0.2.4) I tried 3 or 4 times to upgrade though the Eve app, only to find out that the Sonoff was still on the current version. Every time I gave the command to update I closed both Eve and Home app and waited patiently until the Sonoff was responsive again in home app. But the Sonoff was still in its current version. Eventually the update happened, but I am pretty sure it took more than 3 tries for it to happen.

RavenSystem, 2018-08-09:
@peros550 Many thanks for your feedback. If you want, you can open an issue at OTA repo: https://github.com/HomeACcessoryKid/life-cycle-manager

I really can not do anything, because OTA is another firmware. Your device has 2 firmwares, one is Sonoff_ravencore.bin, and the other is mainota.bin, installed in the second part of the flash memory. When you activate OTA trigger in ElgatoEVE, device simply reboot and load the OTA firmware, that performs the upgrade process, that is completely out of my hands.

peros550, 2018-08-09:
Thanks, I suspected that. Just wanted to share my experience and possibly ask if anybody else experienced the same thing. 
Have a good day!