RavenSystem/esp-homekit-devices #27: Thermostat

riddik14, 2018-08-08:
Hi I use a SonOff base as thermostat as testing. When I setup in app home impossible working in auto mode. Why? Can I use sonoffbasic as thermostat?

RavenSystem, 2018-08-08:
Auto mode is disabled, because Sonoff doesn't know if you have connected a heater or a cooler to it. Auto mode only makes sense in HVAC systems, and a Sonoff can not manage it.

riddik14, 2018-08-08:
Thermostat is developed for a SonOff dual per cold and heather?

RavenSystem, 2018-08-08:
No, sorry. Because Sonoff Dual requires some modding to add a temp/hum sensor. And main purpose of this firmware is to provide an out-of-the-box functionality.

riddik14, 2018-08-08:
wait, I wanted to understand how the thermostat is configured. I mean to understand which device should be set. I tried it on a sonoff basis and it works, but hot or cold always activates that relay or is designed to activate a relay with hot another with cold?

RavenSystem, 2018-08-08:
You have one relay. If you connect a heater to it, then you must use Heater option in HomeKit App; and if you connect a cooler, then you must use Cooler option in HomeKit App.

If you connect a Heater, and you select Cooler option in HomeKit App, relay will activate when sensor temperature is higher than target temperature, and you don't want that.

Kristian8606, 2018-08-14:
RavenSystem can you add a display to the thermostat?
I could not make it .

RavenSystem, 2018-08-14:
@Kristian8606 How many free GPIOs do you have?

Kristian8606, 2018-08-14:
5 GPIO

RavenSystem, 2018-08-14:
And do you have any kind of display?

Kristian8606, 2018-08-14:
I'm trying to add the ssh 1306 OLED display

RavenSystem, 2018-08-14:
Uff, that display requires a lot of flash size, for fonts and other things. Do you have another display (less complex)?

Kristian8606, 2018-08-14:
Nextion 2.4

Kristian8606, 2018-08-14:
https://github.com/SuperHouse/esp-open-rtos/tree/master/examples/ssd1306_example 
this code is successfully compiled along with an esp8266 thermostat with a ssh1306 display

peros550, 2018-08-22:
@Kristian8606 any luck on adding the ssd1306 display to Ravensystem code?

Kristian8606, 2018-08-22:
Успях да изпратя температурната влажност и целевата температура на дисплея, като отпечатах стойностите. Направих серийна връзка с ардуин с прикачен дисплей, който чете данните и ги показва на дисплея. По-късно ще кача кода в моето хранилище заедно със снимки.

RavenSystem, 2018-08-29:
@Kristian8606 No enough flash space to implement HomeKit framework and SSD1306 display in 512KB.

Kristian8606, 2018-08-30:
RavenSystem thanks!
one of the ways to display the data that I did of is with the second esp8266 via a serial interface.
For now, this option works for me with slight exceptions. Sometimes the temperature disappears from the display and appears when the data arrives again, maybe not read from bufer corectly .

I was thinking now another idea.
Is it possible to insert an example of esp-now or another protocol to send the temperature data via the WIFI to a second esp8266 to display them in the display?

Kristian8606, 2018-08-30:
https://github.com/Kristian8606/Sonoff_TH

[image]
[image]
[image]

RavenSystem, 2018-08-31:
@Kristian8606 I have found this example code for an UDP connection:
- Server (esp with display): https://github.com/willemwouters/ESP8266/wiki/UDP-Server---LWIP-Stack
- Client (esp with Thermostat): https://github.com/willemwouters/ESP8266/wiki/UDP-Client---LWIP-Stack

`char data[]` is what you want to send, and remember to fix length in `os_memcpy (b->payload, data, ???);` according to data[].

You must send all data every something change, so you must call send data function when temp change and when you change the state of thermostat.

To send data you must create a simple protocol, so you simply can do this:
char[0] = current_temperature
char[1] = target_temperature
char[2] = humidity
char[3] = current_state
char[4] = target_state

When server receive data, you refresh your display with all data received.

I hope this can help you.

maccoylton, 2018-09-01:
I have the SSD1306 display working with my thermostat example 

https://github.com/maccoylton/homekit_wifi_thermostat

file size is 447904 according to MacOS. Characters are working, my bitmap logo isn't working yet

Kristian8606, 2018-09-01:
if I correctly understand you have an example of a thermostat with a display? Can you give more information about how it was done? sorry my bad english

maccoylton, 2018-09-01:
The SSD1306 is connected directly to the ESP8622, just needs two pins, uses the driver provided as part of esp-open-rtos, my firmware usee OTA, so < 512k, checkout the code at the link above

Kristian8606, 2018-09-16:
maccoylton  I tried your code but when I connect a display and esp starts to restart?

Kristian8606, 2018-09-16:
I use esp8266. when I start the device I do not detect it and I can not connect it to wifi but in the serial monitor it prints temperature. If I connect a display to it I see WiFi but the device is restarted continuously. the display is ssd1306

maccoylton, 2018-09-18:
Did you install my firmware or did you write you own code? How exactly have you word it up? What power supply are you using and what is it rated to?

Kristian8606, 2018-09-18:
I installed your firmware. the same esp 12 works well with the of maximkulkin thermostat.

maccoylton, 2018-09-18:
What version of SS1306 do you have i2C or SPI, the firmware is written for I2C, did you check the code for the GPIOs and have you got a power supply suitable to power the ESP, Temperature sensor and screen?

Kristian8606, 2018-09-18:
I2c display power supply is 2 amps gpio  are connected correctly. When I upload another esp demo code, it all works.

maccoylton, 2018-09-22:
Can you provide more detail please, specific on wiring, can you also provide a trace, per above this is running stable for me, I can help but I need more details.

Kristian8606, 2018-10-09:
The connection is on this circuit pin 4 and pin 5 also tried with pin 14 and 5 volt supply to the display but the effect is the same.
esp is restarted.

[image]

Kristian8606, 2018-10-09:
maybe the display does not cause the problem.
esp is restarted in the same way not only when the display is connected a and when I connect esp to Wifi.

maccoylton, 2018-10-12:
Can you provide a trace please

stepanyurkiv, 2018-11-05:
Сan you use a temperature sensor DS18B20?

RavenSystem, 2018-11-05:
esp-open-rtos includes drivers for DS18B20, so you can use it in your custom firmware: https://github.com/SuperHouse/esp-open-rtos/tree/master/extras/ds18b20

yusufd1, 2018-11-13:
> esp-open-rtos includes drivers for DS18B20, so you can use it in your custom firmware: https://github.com/SuperHouse/esp-open-rtos/tree/master/extras/ds18b20

Are you able to add this as an option in Ravencore that one can select like the other temp/humidity sensors. I want to also use this sensor as I want to deploy it in my swimming pool and it is the only one made to be used submersed. Or maybe you can guide me on how to put this into my custom firmware unfortunately I am not too good at coding.

RavenSystem, 2018-11-13:
@yusufd1 have you a ds18b20 sensor (I have not one)? In that case, open a new issue about that, so we can do some testing.

faronov, 2018-11-14:
@RavenSystem I also think that adding ds18b20 sensor is a good idea 
I also want to use it as a water heater for aquarium 
Also, I see that Apple added Water Heater characteristic, it will be also good idea add this