RavenSystem/esp-homekit-devices #28: SonOff dual multipress

riddik14, 2018-08-08:
Can you add multipress function on SonOff dual? Your fw work fine. Thanks u for your job

RavenSystem, 2018-08-08:
Sonoff Dual has only one button. With a single press, relay 1 is activated, and with a double press, relay 2. I haven't a Sonoff Dual unit, so I can not test it.

riddik14, 2018-08-08:
I have a sonoff dual configured with tasmota with 2 button, 1 for rele. It work with tasmota

RavenSystem, 2018-08-08:
Sorry, I don't understand what you say exactly.

riddik14, 2018-08-08:
I have a sonoff-dual with the ignition of two separate buttons connected to the button1 and button2 pins. in addition to this on every button I have the function of double pressure that activates scenes through mqtt and homebridge with the statelessswitch plugin.  This with tasmota fw

RavenSystem, 2018-08-08:
Sonoff Dual comes with only one button, that RavenCore uses to manage both relays (Single and double press). Of course, you can add additional buttons and sensors, but for now there is no plans to support those functionalities.

RavenSystem, 2018-08-29:
Now, with version 0.3.0, additional buttons (connected to the button1 and button2 pins) work, controlling relay 1 and 2.

closed 2018-08-29 by RavenSystem