RavenSystem/esp-homekit-devices #30: Add pressure/temp sensor

riddik14, 2018-08-08:
Can you add bmp085 bmp180 sensor.

RavenSystem, 2018-08-08:
Apple HomeKit have not support for pressure sensor. I could be added using third party HomeKit Apps, but it will be a new accessory.

riddik14, 2018-08-08:
But from eve is read

RavenSystem, 2018-08-08:
Yes, as I have said, using a third party App (like Elgato EVE). I marked this issue as enhancement, so it is possible to add support to pressure sensor in a future, but my time for coding is limited, and there are others features that I want to add first, such as garage door opener. I will buy a pressure sensor, but not for now.

RavenSystem, 2018-08-09:
UUIDs for Elgato EVE devices: https://github.com/ebaauw/homebridge-lib/blob/master/lib/EveHomeKitTypes.js

RavenSystem, 2018-08-09:
@riddik14 Have you any pressure sensor?

riddik14, 2018-08-16:
Bmp085

RavenSystem, 2018-08-29:
I am waiting a BMP280 sensor.

RavenSystem, 2018-11-01:
Actually I only support services supported by Apple HomeKit, and pressure sensor don't make sense.

closed 2018-11-01 by RavenSystem