RavenSystem/esp-homekit-devices #33: Probably Interference

riddik14, 2018-08-09:
I state that I was using the tasmota firmware. I loaded the firmware ravencore ota on 3 sonoff basic. after having paired from a home app, if you turn on one of them, all three of them are ready. I do not have automations, so I think only of interference. How do I solve? do I put a capacitor on L and N? use normal light bulbs.

RavenSystem, 2018-08-19:
Try disabling GPIO14 in Elgato app.

riddik14, 2018-08-19:
To do. Second me is a rtos problem. The software di in lock. I have to try to insert a condenser on rele on gpio 14 and gpio0 but the problem is persistent. With tasmota work good. There is the problem only when 220v is connected. Why?

RavenSystem, 2018-08-19:
If you want to use GPIO14, you must insert a capacitor between GPIO14 and GROUND to avoid 220VAC interferences. I know that issues doesn’t happen with Tasmota, but it happens with esp-open-rtos. I included a software filter over GPIO14 to avoid interferences, but on some cases, is neccessary to use a capacitor too. 

Does your issue happen with GPIO14 enabled only?

riddik14, 2018-08-19:
with device 3 and 6 and condenser on GP14 the sonoff does not read the dht. device 1 works but sometimes goes to lock or I press the button and the relay turns on for the time I hold down, otherwise it goes off. maybe the capacitance of the capacitors is wrong, do you suggest me the appropriate capacitors?

RavenSystem, 2018-08-19:
Wait, wait. GPIO14 capacitor is only necessary for device type 1, and only if you use the GPIO14 toggle. Please, remove all unnecessary capacitors and explain issue from the beginning indicating details about device types used.

riddik14, 2018-08-19:
I tried your firmware on three basic sonoffs. 1 - light + dht22 (device 6);
2 - light + dht22 (device6);
3 - light (device 1).
when I turn on 3 I also turn on 1 and 2.
when I turn on 1 they turn on randomly, but once I release the button the button (gpio0) goes off and goes to lock.
I tried to put a capacitor on the relay in parallel to the zener diode of the coil contacts to eliminate the disturbances on all three sonoffs, as is done for finders 26.02. despite that still interference but less on sonoff 2 and sonoff 3 when I turn sonoff 1. then I also put on gpio0 sonoff2 and sonoff3 a condenser, they go better but sometimes stop. the sonoff1 stops anyway despite the capacitors. I'm trying to filter out all the interference. I also tried to put a 0.01nF xt capacitor on input 220 but I can not filter out the interference.
all this does not access with tasmota.

RavenSystem, 2018-11-01:
Have you tried latest RavenCore release (v0.4.3)?

closed 2018-11-04 by RavenSystem