RavenSystem/esp-homekit-devices #34: Electrodragon Support ?

Codi878, 2018-08-14:
Hello

First thank you for your hard work. This is amazing and my sonoffs working very good on 2.4 firmware.
I have request please add support for electrodragon. 

https://www.electrodragon.com/product/wifi-iot-relay-board-based-esp8266/

I am test your firmware but only one off two switch work good.

Thank you 👍

RavenSystem, 2018-08-19:
Have you an Electrodragon board? Please, can you provide info about GPIOs used?

Codi878, 2018-08-24:
Yes I have electrodragon board. 

I found such a scheme where the inputs and outputs are nicely described.

is it enough if you need more information?

[image]

Codi878, 2018-08-24:
[image]

RavenSystem, 2018-11-26:
Added to TODO List: 
https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List