RavenSystem/esp-homekit-devices #35: Set initial state to no.

gooring, 2018-08-18:
Hello there. Firstly I have to say thank you very much for your work. The Homekit and Eve integration is a great idea and the flashing went flawlessly.

I just have one question. How do I set it up so the initial state of the Sonoff basic is on. For example, if I have a Sonoff between a wall light switch and a LED light, when I turn on the switch I want the LED to turn on right away. Sorry for the stupid question. I'm very new to programing. Thanks

RavenSystem, 2018-08-18:
A custom option will let you to setup initial state:
0. Off (default)
1. On
2. Last state
3. Oposite to last state

gooring, 2018-08-19:
Thank for the Reply. Do I have to do this in the main.c file? Thanks

RavenSystem, 2018-08-19:
You must modify main.c, and custom_characteristics.h to declare new HomeKit UUIDs to customize initial states.

gooring, 2018-08-19:
I don’t really understand. I think I have a lot to learn before I get to do it. Anyway, thank you very much.

peros550, 2018-08-25:
@RavenSystem Hi Jose, if you could implement such an enhancement it would be extremely useful. This function could have many use cases especially in areas where electricity has temporary disruptions, etc. Having the option to set the output to last state would overcome such issues and automations would not be disrupted

RavenSystem, 2018-08-25:
@peros550 This will be available asap. I need enough free time for coding. ;)

RavenSystem, 2018-08-25:
There is a little bug with GPIO14 and Basic Switch. Initial state must be OFF, but depending of position of wall-mount switch connected to GPIO14, some time initial state is ON.

This bug must be fixed before implementing saved states.

RavenSystem, 2018-08-25:
The bug with GPIO14 initial state for Device Type 1 has been fixed in v0.2.6, released just now.

Next step will be implementation of "Setup initial state".

Argyropus, 2018-09-13:
is there any news? I'm really looking foward to this.

aceyyoung, 2018-09-17:
I am really looking forward hearing the new about the initial setting too. Some light switchs should be set to be initially on, as we can control the light status with the wall switch and with homekit at the same time. Thank you very much for your hard work as always.

aceyyoung, 2018-09-17:
is it possible to put this function into setup menu then we can configure it according to the usage?

RavenSystem, 2018-11-04:
Added to TODO list: https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List

closed 2018-11-04 by RavenSystem

RavenSystem, 2018-11-27:
Custom initial state has been added to version 0.5.0.