RavenSystem/esp-homekit-devices #36: Plug when WiFi is not yet available, stays in unresponsive homekit state.

ageorgios, 2018-08-22:
When I plug in the sonoff device while the WiFi is not available yet, the state stays unresponsive even though the WiFi becomes available later.

RavenSystem, 2018-08-22:
Sonoff RavenCore 0.2.4 keeps searching wifi for 75 secs, then it turns into AP mode. Wifi must be available into first 75 secs since device boots.

ageorgios, 2018-08-22:
This means that my FritzBox 7490 router needs more than 75sec... please consider doubling the timeout on a next release

Thanx.

RavenSystem, 2018-08-24:
Try version 0.2.5.

ageorgios, 2018-08-25:
awesome thanks!

peros550, 2018-10-02:
Hi Jose, 

in latest version 3.0.0 I still have this issue. My router is an ASUS RT-AC68U and it takes a lot to boot and activate wifi. Would it be possible to implement it in a way that if no client is connected in AP mode for 1-2 minutes, it will try again to connect to old known Wifi network?

RavenSystem, 2018-10-02:
Now I have an ASUS ROG Rapture GT-AC5300 and it takes a lot to boot too.

xtianpaiva, 2018-10-02:
Same situation here with Unifi AP in my house, taking too long to boot... probaby the solution is something similar to @peros550 "retry" proposal. Gran trabajo, saludos!

closed 2018-11-01 by RavenSystem