RavenSystem/esp-homekit-devices #37: Water Valve set schedule

mmiglier, 2018-08-24:
Hi, thanks for the awesome project. Is there any plan to introduce the ability to set a schedule on water valve like the Elgato Eve app does for the Aqua Eve? I’m not sure whether the homekit sdk allows this or even the esp allow to keep the calendar time and stay in sync, I’m pretty new to this world. If you see any possibility and you would like to give some hint I would happy to contribute.

RavenSystem, 2018-08-24:
Sorry, but HomeKit specs only provide to set duration. But if you want to set schedule, you can use an AppleTV or iPad as accessory hub, and you can create an automatization. I use that with a Sonoff Basic and a 220VAC electro valve.

mmiglier, 2018-08-24:
I see, that’s what I feared. That’s the solution I adopted. I created two scenes from Elgato Eve App. One for setting the default time duration to 15 minutes at 6:59 and one to start the irrigation at 7:00. It is a workaround though. I hope they’ll improve it soon.

closed 2018-08-24 by mmiglier

RavenSystem, 2018-08-24:
You can set the default time directly in accessory. That time is saved in flash memory. 

Schedule implementation like Elgato accessory is very hard, for example, esp8266 has not a clock for date/time.

mmiglier, 2018-08-24:
Yes but unfortunately it always gets replaced by the latest value I used. I prefer to set it to the value I need it to be in the morning.

mmiglier, 2018-08-24:
I see there are ways of setting date/time actually. E.g.: http://pedrominatel.com.br/en/esp8266-en/date-and-time-on-esp8266-using-ntp/