RavenSystem/esp-homekit-devices #39: Sonoff 4ch switches can’t stay in different rooms

mmiglier, 2018-09-03:
Hi,
I noticed the 4 switches can’t stay in different rooms when adding a 4ch. I guess the problem is with homekit protocol, which is identifying the 4 switches as a single accessory. Do you see any viable workaround for such a problem?
Thanks!

RavenSystem, 2018-09-07:
To improve that, it will be necessary to declare each service in a separate HomeKit accessory.

GPL71, 2018-10-07:
@mmiglier 
I think you’re italian .... anyway I can Write english.
Which sonoff 4ch are you using?
4ch or 4ch Pro.
Is firmware compatible with 4ch pro R2? @RavenSystem 
Can I switch on the outputs at the same time? Using oringinal sonoff firmware you can use only 1 output each time
Grazie

mmiglier, 2018-10-07:
I have the 4ch pro r2 and it works perfectly except for the problem I described in this issue. Working modes should be selected using K5 and K6 toggles as described in the manual: http://ewelink.coolkit.cc/?p=458

GPL71, 2018-10-07:
Thank you @mmiglier for your replay.
Then k5,k6,s5 and s6 will work in the same way of original firmware sonoff.
Is that right?
Now about your issue.... I want use the 4CH  R2 in the bed room... 4 different bulb in the same room.
If I understood, I’ll not have problem.
Correct?

mmiglier, 2018-10-07:
I think hardware toggles can not be programmed any different without changing also the circuitry.

As for my issue, that’s correct, it’s not a problem for your use case.

GPL71, 2018-10-08:
Grazie Marco ....
Then ....
1- I'll flash Raven firmware
2- after first settings, using EVE App I'll change device to 4ch pro R2 and set K5 and K6
3- device will restart
4- ready to use....

Grazie ancora

RavenSystem, 2018-11-26:
Due to low RAM resources of esp8266, setting an individual room for each service requires creating an accessory for each switch, and this is not a good idea with an esp8266.

Remember this hierarchy:
Accessories -> Services -> Characteristics
And switches that you see in Home App are services, the characteristic is on/off toggle. The accessory is the group of services that has one or more services, and has characteristics like firmware revision, manufacturer, etc. But you can not separate services in different rooms, only accessories.

closed 2018-11-26 by RavenSystem

RavenSystem, 2019-05-18:
New firmware Home Accessory Architect (aka HAA) supports multiroom feature:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect