RavenSystem/esp-homekit-devices #40: Elgato Eve customization

faronov, 2018-09-07:
I install RavenCore firmware
everything is fine 
have some trouble when flashing, but this happens because I use the old version of esptool, and seems it incorrect erase flash.
but now everything is fine

but one question regarding customization
I can't find anything that you describe in a wiki inside Elgato EVE
Must I add something more? Where I can find these settings
As described in this section of a wiki
https://github.com/RavenSystem/esp-homekit-devices/wiki/Sonoff-RavenCore#user-settings

RavenSystem, 2018-09-07:
You should be able to see an option called "Show setup" inside main accessory characteristic.

faronov, 2018-09-08:
I only see

faronov, 2018-09-08:
Sorry 
I found where this
it would be nice if wiki has details information with pictures :)

closed 2018-09-08 by faronov

RavenSystem, 2018-09-08:
I know it. Pictures are on the way. I created a repo for media to include photos. SO, I need some free time to work on it.