RavenSystem/esp-homekit-devices #43: Raven core v0.3 about the device type 8

sunwon2018, 2018-09-14:
I like this program very much. It’s really useful. 
I am trying to control my garage door. I saw the device type 8 . Even you said it’s still not  complete. I can Trun on the door but just HomeKit displays  it’s opneing and looks like waiting a status returned. 
Could you please tell me about these gpio4 gpio5 gpio14 function. How to connect them so that esp8266 can get a status of the door? 
 ThankS

RavenSystem, 2018-09-22:
Garage door is not completed. gpio5 and 14 will be used for open and close sensors. gpio4 will not be used.

GPL71, 2018-10-08:
@RavenSystem
Jose garage door it's a wonderful idea.. Wich SONOFF will be compatible? 4CH pro?
Some garage door hasn't switch for open/close position, but use current limiter (in italy we have this model too)
Then is possible set a timer for open and close operation?

Thank you for this wonderful work

GPL71, 2018-10-08:
I'm sorry I'd forgot
Is it possible manage rollers?

Thank you again

RavenSystem, 2018-10-21:
Garage Door update has been released.

closed 2018-10-21 by RavenSystem