RavenSystem/esp-homekit-devices #46: Stability during OTA setup and pairing of more sonoff

GPL71, 2018-10-02:
Hello! Great work!! All is working fine.

FIRST QUESTION
I want report only a stability problem during setup operantios.
The Apple device can connect to SONOFF and setup starting.
It seem you've a very short period to complete setup, about 1 minute, sometimes less.
After that time if you couldn't complete it, SONOFF turn in reset (?) and you must restart process connecting toAP wifi and wait the web interface and restart ....and so for several times.
Is there a way to fix problem or a work around?

I have 2 SONOFF and both are affected by same problem, and even changing device (at least 4) the problem is present.

SECOND QUESTION
The pin code for pairing of device is in the firmware.
Now I would program several SONOFF in my home.
How do I do?
Is it necessary compile a new firmware .bin for each device?
Can I find a guide or can you explain how to?
I'm using MAC oSX. Is there a way to read the .bin file and modify the pairing?

Be patient ... I'm not a programmer. I'm learning 
Thank you

peros550, 2018-10-02:
Jose, 

taking the opportunity I would like to confirm that latest firmware works great. 

In regards to the OP questions here are my comments. 

a) That's a valid question. If you are a little late in typing the passwords, or even in the very first OTA stage  where you have to manual give the github repo and the bin, then there is a big chance you get delayed and system time-outs. 

b) I have used at least two sonoffs with the same pairing code and didn't have any issues. However, it would be very easy to remember if there was a much simpler code (i.e. 111-11-111)

GPL71, 2018-10-02:
Thank you very much for reply ....

About 1st question
I did this work around
before I'd typed the .bin file, after repository, and last the wifi network ...

About second question
I've no way to pairing the second sonoff with the default pin you'd set.
How can I resolve?

Is it possible EDIT your .bin file to change pin?
If so how?

I would like about 10 sonoff in my home...

THIRD Question

can you add the function Button switch to the relay . May be programmable ( 1 to 3 second)

Last ... When garage door will be available?

Thank you

RavenSystem, 2018-10-22:
1st) I suffer that wifi-config issue too, but i'ts a problem that must be solved in wifi-config repo; but I think that it's an iOS timeout and there is nothing to do for fix it.

2nd) I have paired many RavenCore devices with the same pin. Actually I have 7 RavenCore devices working in my house (1x Basic, 1x S20, 1x S22 w/TH sensor, 1x SV and 3x TH16 w/sensor) . A friend has 12 Sonoff Basic with RavenCore working in his house.

I don't know if it's possible to edit the bin file, but for sure if you edit it, OTA updates will not work.

3rd) I only incorporate features supported by Apple HomeKit, and a push button service is not supported.

Last) Garage door is available now. See Wiki for setup instructions.

closed 2018-11-01 by RavenSystem