RavenSystem/esp-homekit-devices #50: Suggestions about Firmware

GPL71, 2018-10-05:
Before, I want confirm this is a very good work! It work fine! Fantastic!! Best work actually available

I would like suggest following implementations:

1 - Pairing PIN code customizable in EVE App. 
      First time you join device using default code, then by EVE App you customize pairing PIN.
      Then restart device to set new code in the device, and pairing again.
      These allow use of multiple units of SONOFF and manage them

2 -  Function temporized button. This allow open garage door that use only 1 contatct or elctro lock
       or  generally a temporized closing.
       The time would be customizable by steps of 0,5 sec

3 - Function water leak. If a leak is detected then sonoff energize to set on an electrovalve and close 
      water flow

I hope you'll consider these implementations.
Thank you very much for your attention

RavenSystem, 2018-10-05:
1. You can add many devices using same HomeKit code. Actually, I have 6 devices using RavenCore. 

2. It can be possible, but a complete and customizable garage door solution is on the way.

3. It can be a new device type: water valve + leak detector

Many thanks for your suggestions.

GPL71, 2018-10-08:
Thank you Jose for your replay and because you keep in consideration these suggestions.
I want remark the firmware is wonderful and stability is fantastic.
About point 1... Pairing code. After several attemps HOME APP paired the second SONOFF, but don't ask me steps (LOL).. 
If I set a second house pairing is quicly (but this a workaround)
However now the second SONOFF is paired in the same room and obviously same house.
Then if you can add the feature to change the pin code by EVE APP it would be fantastic!!

Another on request... (I'll stress you with ideas because this project is wonderful)
Can you add RGB strips control?

Thank you for you attention ....

i3laze, 2018-10-12:
+1 for water valve class.
There are lot's of 220V electric valves for any pipe diameters on AliExpress. Will get one someday.

RavenSystem, 2018-11-10:
Some notes:
- RGB lights will be supported in a future, but not soon (It's in TODO list).
- PIN code will never be changed. It will be 02182017 forever. (Of course, you can compile firmware by yourself and change it).
- Sonoff 4CH in valve mode will never happen. Manage 4 individual timers with an esp8266 and HomeKit running is snot a good idea. Buying 4 Sonoff Basic is the better solution.

closed 2018-11-10 by RavenSystem

xrust83, 2020-12-16:
02182017 looks like a date. She's probably very important to you