RavenSystem/esp-homekit-devices #52: OTA-Update loop

rossiniscarface, 2018-10-07:
Hi,

first of all: Very nice work. Thank you for your firmware i'm using it in one Sonoff Basic.

I had a problem with the OTA-Update: After initializing it with the Eve App, the Sonoff Basic never rebooted.
On my router i saw that the sonoff connected to some server with names like lb-192-30-253-124-iad.github.com:443 and loaded around 4-40KB data.
Then nothing. 
I let it there for 30 minutes but it never rebooted in the process. On the router the connections from the sonoff repeated like this:
dhcp request to router
dns request to router
https request to lb-192-30-253-124-iad.github.com:443 or similar
and after a while it repeats with dhcp request.

After 30 minutes i cut the power and the sonoff was available again but with the old fw-version 0.3.
Just for fun i changed it in the eve app from device type 1 to device type 8 and rebooted and tried the firmware update again.

Now i saw on the router that the sonoff connected to a generic amazon-server and loaded around 600KB.
After a short time it rebooted and was available in HomeKit with the new firmware-version 0.3.2

On my first install over serial-port i had a similar experience and there i could see that the github url was correct, as my computer could download the file, but in the serial-connection it state a 404.

RavenSystem, 2018-10-08:
OTA fails sometimes, or many times. When I update using OTA, I wait 6 minutes before reset device and try again (and sometimes again, and again...). Since OTA system is not my work, you can open an issue here: 
https://github.com/HomeACcessoryKid/life-cycle-manager

rossiniscarface, 2018-10-08:
Well, thanks for your info. Now i know i am not alone. 
Would it be possible to somehow inform the user about the update process.
e.g. Start Update --> Blink 3 times short on gpio0
Finished Update --> Blink 3 Times long on gpio0
Failed Update --> etc ...

riddik14, 2018-10-08:
In serial consolle you can monitorized update process

peros550, 2018-10-11:
In case there is an update to the OTA firmware on git, how do we get it?

RavenSystem, 2018-10-12:
When you enable Update Firmware trigger, OTA firmware search updates for itself too.

peros550, 2018-10-12:
that's great!

closed 2018-10-22 by RavenSystem

HomeACcessoryKid, 2018-11-17:
note that I have the intention to fix the issue. See the [LCM](https://github.com/HomeACcessoryKid/ota/issues/3) issue.