RavenSystem/esp-homekit-devices #53: Issue #26 reappeared

peros550, 2018-10-10:
Hi Jose, 

with latest update I the same issue as discussed here: 

https://github.com/RavenSystem/esp-homekit-devices/issues/26

Can you please check?

RavenSystem, 2018-10-10:
Please, can you test this beta version?

[sonoff_ravencore_0.3.3beta1.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/2464106/sonoff_ravencore_0.3.3beta1.bin.zip)

peros550, 2018-10-10:
Sure, is it possible to install it though OTA?
Otherwise it would need to open the electrical panel :(

RavenSystem, 2018-10-10:
I have released v0.3.3 to the public just now. Try it using OTA.

peros550, 2018-10-10:
Will do that in about 4-5 hours when I get home. 
Many thanks for such a quick response!

peros550, 2018-10-10:
I just tested 0.3.3. There is an improvement for sure, but not completely bug free. Here is what I mean. In 0.3.2 I had random resets to AP mode. Those seem to have been eliminated. 

However, when I try to either control the output through the Home App, some of my commands get rejected. Have a look at the video.

https://www.youtube.com/edit?o=U&video_id=vs7wUfz3A2c

Thanks,

Codi878, 2018-10-10:
Hi Jose,

Yes, I am confirming this error on 3.3 firmware. Version 3.0 worked more stable according to my assessment.

Best Regards

RavenSystem, 2018-10-10:
[Sonoff_RavenCore_0.3.4beta1.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/2466405/Sonoff_RavenCore_0.3.4beta1.bin.zip)

Codi878, 2018-10-10:
Hi Jose,

Should I upload it instead of the otaboot.bin file?

Best Regards

RavenSystem, 2018-10-10:
@Codi878 Yes!

Main problem is that I'm not able to reproduce @peros550 YouTube issue, but I know the cause of the issue.

Codi878, 2018-10-10:
Ok 

> Wiadomość napisana przez José A. Jiménez <notifications@github.com> w dniu 10.10.2018, o godz. 23:07:
> 
> Main problem is that I'm not able to reproduce @peros550 YouTube issue, but I know the cause of the issue.

Codi878, 2018-10-10:
Jose,

I have tested new firmware and It looks like it works right now. I mounted it in the same place - the pump controls the inductive load and now responds to my attempts to turn off. 

Earlier I clicked and Sonoff returned to its previous state just like in the demonstration film above from @peros550 

Thank You for quick response :) and great job.

RavenSystem, 2018-10-11:
Many thanks. Version 0.3.4 has been released just now.

peros550, 2018-10-11:
I can confirm, the issue was resolved. By the way what caused this issue?

RavenSystem, 2018-10-12:
@peros550 Main cause is that I have recoded all button functions and now them use a new library that I coded. Advanced Button Library. Firmware code was growing in a way that need a reorder to continue the development in an easy way. Now it's very easy to implement different callback functions to each button press type. 

Advanced Button Library: https://github.com/RavenSystem/esp-homekit-devices/tree/master/libs/adv_button

RavenSystem, 2018-10-22:
@peros550 @Codi878 Please, can you update to last version (0.4.1) via OTA and check if issue reappears? Now I use hardware timers in Advanced Button Library for better stability and performance.

peros550, 2018-10-23:
Hi ,sorry to bring bad news, issue reappeared in 0.4.1.

RavenSystem, 2018-10-23:
@peros550 many thanks. I’m going to implement the system that deactivate buttons while relay is working.

RavenSystem, 2018-10-23:
@peros550 Please, update to version 0.4.2 via OTA. Many thanks!

peros550, 2018-10-23:
It works fine again! Should you want me to do some testing again let me know , preferably weekdays afternoon.

RavenSystem, 2018-10-23:
@peros550 Many thanks for your help!!!

closed 2018-10-23 by RavenSystem

peros550, 2020-01-05:
In the early days of Ravencore we had found a strange issue when Sonoff works very close to high Amp relays & lines. If I am correct, you coded a filtering function to overcome those issues. Is there some special configuration applied on the JSON for such case?

I have a sonoff basic which is going to be upgraded in HAA (one of the last few). 

Thanks & Regards