RavenSystem/esp-homekit-devices #54: Hi, 1 of 3 Sonoffs stuck updating from 0.3.2 to 0.3.4, and suspicious WiFi Channel change behaviour...

Open-Elec-RU, 2018-10-12:
Hi - great work!

Successfully updated one Basic from 0.3 to 0.3.2 using the Eve app.

Then had to change the 2.4 channel on one of the Airport Extremes that this Basic's was connected to.  Ended up in AP mode.  Coincidence? Just in case it was the _rebooting_ of the Airport Extreme that triggered this, restarted it one more time, but the Basic reconnected just fine.

Then today, updated two other Basics from 0.3 straight to 0.3.4 without ant problem. 

Trying to update that first Basic from 0.3.2 to 0.3.4 gave me an unresponsive unit, no AP mode, no response.  Had to erased/reflash it again - we'll see if it's working...

(Other than the occasional hitch above) this is really cool!

Had to report! :)

https://www.reddit.com/r/HomeKit/comments/9gnas7/custom_sonoff_firmware_with_extra_features_and/

RavenSystem, 2018-10-15:
Many thanks for your feedback.

Sometimes, device keep in OTA process (no AP mode and no response). If you push device button and LED doesn't turn on, device is in OTA, and the only way without refresh is to wait 5 minutes, power off and on again, and wait 5 minutes, doing this until device button turn on led again, indicating that device has loaded RavenCore firmware instead OTA firmware. I had this issue a few times.

RavenSystem, 2018-10-21:
Added to Wiki.