RavenSystem/esp-homekit-devices #55: Does this work with the newer Sonoff S26 (two-six)?

Open-Elec-RU, 2018-10-14:
I know they are physically somewhat harder to flash (tx/rx/vcc/gnd) are all over the board, but is it compatible with RavenSystem firmware?

Thanks

RavenSystem, 2018-10-14:
I don’t know, but RavenCore could work using device type 3, like a S20.

RavenSystem, 2018-11-01:
@Open-Elec-RU Do you have a S26? Have you tried RavenCore with it?

Open-Elec-RU, 2018-11-02:
¡Hola!
Not yet, just ordered some - will report! (takes forever from aliexpress, though)

RavenSystem, 2018-11-02:
@Open-Elec-RU Many thanks!!

RavenSystem, 2018-11-02:
@Open-Elec-RU I have read it. The main problem is that I'm using an old version of esptool (included in esp-open-sdk to develop) under MacOS. If you can help me a little with your experience to fix Wiki instruction...

1. when doing the first part in Step 3, erase_flash, you specify the port name WITHOUT preceding it with "/dev/" (just "tty.usbserial....")
What command have you launched exactly?

2. but when doing the second part, ie actually flashing it, i had to specify "/dev/" ("/dev/tty.usbserial.....")
What command have you launched exactly?

3. you also specify "-fs 1MB", with the "B", not "1M" as the instructions say, otherwise esptools gives you an error.
I understand that correct way is "-fs 1MB", right?

Many thanks.

Open-Elec-RU, 2018-11-03:
I have esptools v2.3.1.
1. Sorry, my mistake - "/dev/" **should** be included. 
2. esptool.py -p /dev/tty.usbserial-A400fq9h --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin
3. Yes

Open-Elec-RU, 2018-11-03:
using "1M" gives you this:
_esptool write_flash: error: argument --flash_size/-fs: 1M is not a known flash size. Known sizes: 16MB, detect, 256KB, 2MB-c1, 8MB, 2MB, 4MB, 512KB, 1MB, 4MB-c1_

RavenSystem, 2018-11-03:
I have made some changes in Wiki. Please, check them.

RavenSystem, 2018-11-03:
> by the way, did you know?
> https://www.reddit.com/r/HomeKit/comments/9q24m7/itead_products_sonoff_to_receive_homekit/
> the latest from this source is "maybe next month" is now "january" :)

In case that iTEAD releases a HomeKit firmware for Sonoffs, that firmware only will expose Sonoff in form of simple switches, without additional functions. For example, with RavenCore you can use GPIO14 to connect an external mount-wall switch in a Sonoff Basic, or control a garage door, or a water valve... If iTEAD make Sonoff HomeKit, I will continue developing RavenCore (My garage door will use it, for example).

Open-Elec-RU, 2018-11-04:
Absolutely! We hope you do!  Your stuff works, while they’ve been promising since April and we’ll definitely be needing another HUB.

RavenSystem, 2018-11-04:
S26 added to TODO list: https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List

danieljcooper, 2018-11-07:
Hi, I have been using esp-HomeKit-devices on 4 Sonoff S26 UK models for over a month now and not had any issues. I don’t believe you’ll need to make any changes.

RavenSystem, 2018-11-07:
@danieljcooper Many thanks!

closed 2018-11-07 by RavenSystem

xrust83, 2020-12-16:
I am flashing on Windows 7. I put the firmware files in the folder c:/windows/system32, the programmer is connected to COM7

-=-BACKUP-=-
esptool.py --port COM7 read_flash 0x00000 0x100000 esp8266-backup.bin

-=-ERASE-=-
esptool.py --port COM7 erase_flash

-=-FLASH-=-
esptool.py --port COM7 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 fullhaaboot.bin