RavenSystem/esp-homekit-devices #59: GPIO 14 as momentary switch

Fernyman79, 2018-10-23:
I'm trying to use a Sonoff SV with a touch sensor connected to GPIO 14. The sensor works, but it seems that the current implementation is for a toggle, or a mechanical switch, so as soon as I stop touching the sensor, the Sonoff (switch) turns off. The onboard button acts as a momentary switch, but I couldn't find a way to wire the sensor to it's contacts. Any idea on how this can be achieved?

RavenSystem, 2018-10-23:
Yes, GPIO14 in dev type 1 is for a toggle. I can add a setup option to select it for a touch button.

Fernyman79, 2018-10-23:
That would be awesome!

I'm really enjoying your work on this project, btw. It finally made HomeKit viable to me. I can't really contribute, since I'm not an experienced programmer, but I'd love to support it on any other way.

Fernyman79, 2018-10-30:
Curious, is this this something I can change in the source and compile myself? I tried to find any reference in the code, but I'm not sure what I'm looking for.

RavenSystem, 2018-10-30:
It's a new feature that must be developed and implemented.

RavenSystem, 2018-11-04:
Version 0.4.4 changes the way to manage GPIO14 in dev type 1. If new way works fine to avoid AC interferences, I will can add an option to setup GPIO 14 as momentary switch easily.

RavenSystem, 2018-11-04:
Added to TODO list: https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List

closed 2018-11-04 by RavenSystem

RavenSystem, 2018-11-12:
Ready and tested!

https://github.com/RavenSystem/esp-homekit-devices/releases/tag/0.4.7
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#gpio14-toggle-default-off