RavenSystem/esp-homekit-devices #60: PWM dimmer. Single color led strip or cct ”dual color”

jukorp, 2018-10-23:
can you help me?
I need pwm dimmer. ”One color led strip”

-One button. ”Long press to dimming and short press on/off”
-Initial state is off.
-Remember dimmable settings.

RavenSystem, 2018-10-23:
Sorry. This repo is focused on Sonoff devices, but not in custom prototypes. Maybe you should do your question is this repo: 
https://github.com/maximkulkin/esp-homekit-demo

Kristian8606, 2018-10-23:
This is the same
https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/magic_home_strip/magic_home.c

jukorp, 2018-10-29:
Can you check this?
This is sonoff product.
https://www.itead.cc/sonoff-led.html

Includes a features Cct and rgb.

Kristian8606, 2018-12-27:
I have a similar device going i give the code when it's completely done

Kristian8606, 2018-12-27:
https://github.com/Kristian8606/simple_dimming_LED

homkittoby, 2018-12-29:
@Kristian8606 

Hi,
Did you also write a version with bin file for the original MagicHome LED strip controller?
#define RED_PWM_PIN 5
#define GREEN_PWM_PIN 12
#define BLUE_PWM_PIN 13
greetings

Kristian8606, 2018-12-30:
My English is very bad, and I'm not sure I understand your question properly

closed 2019-01-27 by RavenSystem