RavenSystem/esp-homekit-devices #62: How to convert file main.c to .bin?

longttcd, 2018-10-24:
I downloaded the device source esp-homekit, I want to load for each device specific device directory, not the file revancore.bin. But how do I find .bin files for each device that I only see in each device. C. How do i convert from. C to .bin to load as sonoff_revancore.bin?
Thank you

RavenSystem, 2018-10-24:
In the Wiki are the build instructions:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions

longttcd, 2018-10-27:
I'm a beginner, can you guide me step by step?
I tried to follow the above instructions but not completed. I tried to do in the linux command but also did not complete.
I want to customize gipo for Sonoff T1 US 1 - 2 - 3 Gang.
Thank you

RavenSystem, 2018-11-08:
T1 - 1 and 2 gangs should work with dev type 1 and 2. Touch switch with 3 can be supported if you provide me info about used GPIOs.

guagsak, 2018-11-09:
Here it is from tasmota code for Sonoff T1 3G US

```
 { "Sonoff T1 3CH",   // Sonoff T1 3CH (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,        // GPIO02 Optional Sensor (J3 Pin 5)
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_REL3,        // GPIO04 Blue Led and Relay 3 (0 = Off, 1 = On)
     GPIO_REL2,        // GPIO05 Blue Led and Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_KEY2,        // GPIO09 Button 2
     GPIO_KEY3,        // GPIO10 Button 3
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Blue Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  }
```

RavenSystem, 2018-11-09:
Added to TODO list.

RavenSystem, 2018-11-17:
Update to version 0.4.9, please.