RavenSystem/esp-homekit-devices #63: AP mode after hard reset

Codi878, 2018-10-25:
Hi Jose,

After loading the 4.2 firmware and after configuring everything works fine, but when I do a hard reset (10-second button), I can not connect to the sonoff basic network. It only remains to reload the software.

[image]

RavenSystem, 2018-10-25:
Can you provide logs, please?

Can you try with another services?

Try to do a hard reset again, please.

Codi878, 2018-10-25:
Exactly the same problem occurs on two different sonoffs even if the software is loaded without OTA. Unfortunately, I do not know how to read the logs but in my opinion something bad happens during a hard reset and later hard reset again, nothing helps

RavenSystem, 2018-10-25:
Can you try this beta and post feedback, please?

[Sonoff_RavenCore_0.4.3beta1.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/2517045/Sonoff_RavenCore_0.4.3beta1.bin.zip)

gpeter62, 2018-10-26:
I have the same problem, here is the log.
After reset always tries to reconnect to the old network and not possible to connect the AP and get the setting web page.

>>> Resetting device to factory defaults
>>> Resetting RavenCore settings
>>> Restarting device
reconnect
>>> Toggle Switch manual
>>> Relay 1 -> 1
scandone
no farm32 found, reconnect after 1s
reconnect
scandone
del if0
bcn 0
del if1
usl
sul 0 0

ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2292, room 16

0x40100000: _stext at ??:?

tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3fff0f90, prio:14, stack:512
pm_task_hdl : 3fff0880, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4fe0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Oct 23 2018 16:04:55
phy ver: 273, pp ver: 8.3

>>> wifi_config: Initializing WiFi config
>>> wifi_config: wifi_config_station_connect: No configuration found
>>> wifi_config: Starting AP mode
>>> wifi_config: wifi_config_softap_start: Starting AP SSID=RavenCore-0F2CC9
>>> wifi_config: Starting DHCP server
>>> Loading show_setup -> 0
>>> Loading device_type -> 1
>>> Loading device gpio settings for 1
mode : sta(bc:dd:c2:0f:2c:c9) + softAP(be:dd:c2:0f:2c:c9)
add if0
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
>>> wifi_config: Starting DNS server
>>> wifi_config: Staring HTTP server
scandone
scandone
reconnect
scandone
no farm32 found, reconnect after 1s
reconnect
scandone
no farm32 found, reconnect after 1s
[... 183 lines cut ...]
reconnect
station: cc:b0:da:b7:61:35 join, AID = 1
scandone
no farm32 found, reconnect after 1s
reconnect
station: cc:b0:da:b7:61:35 join, AID = 1
scandone
no farm32 found, reconnect after 1s
reconnect
>>> wifi_config: http_task: Got 64 incomming data
!!! wifi_config: Unknown endpoint: GET /connecttest.txt
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 26 incomming data
>>> wifi_config: wifi_config_server_on_message_complete: Unknown endpoint
>>> wifi_config: client_send_redirect: Redirecting to http://192.168.4.1/settings
>>> wifi_config: Client disconnected
scandone
>>> wifi_config: http_task: Got 64 incomming data
!!! wifi_config: Unknown endpoint: GET /connecttest.txt
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 26 incomming data
>>> wifi_config: wifi_config_server_on_message_complete: Unknown endpoint
>>> wifi_config: client_send_redirect: Redirecting to http://192.168.4.1/settings
>>> wifi_config: Client disconnected
scandone

RavenSystem, 2018-10-26:
Try version 0.4.3 and reset to factory default after update.

gpeter62, 2018-10-26:
I tried OTA update in Eve from working 0.4.2. It was not successful. (it is a part of the log)

Now I try a fresh upload + OTA.

RavenSystem/esp-homekit-devices@version:"0.4.3"
--- ota_boot...1
--- ota_compare
0-0,4.3-4.2
4-4,3-2
3-2
--- ota_get_hash
--- ota_get_file_ex
GET /RavenSystem/esp-homekit-devices/releases/download/0.4.3/.sig HTTP/1.1
Host: github.com

--- ota_connect
target IP is 192.30.253.113 bind socket 59852....OK. socket connect to remote....OK. create SSL....OK. set_fd done. SSL to github.com port 443....OK
send request......OK

HTTP/1.1 404 Not Found
Server: GitHub.com
Date: Fri, 26 Oct 2018 16:25:57 GMT
Content-Type: text/html; charset=utf-8
Transfer-Encoding: chunked
Status: 404 Not Found
Cache-Control: no-cache
Vary: X-PJAX
Set-Cookie: has_recent_activity=1; path=/; expires=Fri, 26 Oct 2018 17:25:57 -0000
Set-Cookie: logged_in=no; domain=.github.com; path=/; expires=Tue, 26 Oct 2038 16:25:57 -0000; secure; HttpOnly
Set-Cookie: _gh_sess=bERaQk9QK0xXTkF1L3RoY3BQS1gyWGFUTWdOMUdSVWl4MU1DOGMvUVJBdVBNeUowUUhOckhpNDVpQ29yYmlRRXlMeXRSMFdNTHIxbm84MUNjZjJlcCsvK0NscWhLd2g5c2sza01sTTBQTlQ0NHhsU0JVZm9xSWJUWk5kQ2lYTkwzYWRSVEZKV2lKVldkdWlMaVE1cE8ycHg5SzUvYTVLcXhwbVRrc3VGcm1sb3pzUDY0KzVCdjhoOGZUNERpNnlKZVYvenFoTjdySlFBWW90YWhLSHVKdz09LS1KaURMR0JFeDRWcFZPTE5uZWJwMkN3PT0%3D--fc60db3d8c5cf11968ff3f8896441ec28e636038; path=/; secure; HttpOnly
X-Request-Id: 00292c00-bf99-4b2a-b8dc-39e516d7caa8
Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
X-Frame-Options: deny
X-Content-Type-Options: nosniff
X-XSS-Protection:
HTTP returns 404
--- ota_get_file
--- ota_get_file_ex
GET /RavenSystem/esp-homekit-devices/releases/download/0.4.3/ HTTP/1.1
Host: github.com

--- ota_connect
target IP is 192.30.253.113 bind socket 59853....OK. socket connect to remote....OK. create SSL....OK. set_fd done. SSL to github.com port 443....OK
send request......OK

RavenSystem, 2018-10-26:
@gpeter62 In your log I can see this:
> GET /RavenSystem/esp-homekit-devices/releases/download/0.4.3/.sig HTTP/1.1

indicating that you missed `OTA binary file` field in initial setup. You must erase_flash and install otaboot.bin again.

gpeter62, 2018-10-26:
I dont understand. The original 0.4.2 version was uploaded by OTA. But the upgrade from this 0.4.2 to 0.4.3 by OTA failed... I'm afraid, it is not normal. This version upgrade should be tested again...
-------------------------------------------------------------------------------------------------------------
I made a fresh install. Worked, paired. After reset I could connect again to the network and pair again. I think, this bug is solved. Thank you! :)

gpeter62, 2018-10-26:
Why is so big commend?  Sorry, it is not wanted...

RavenSystem, 2018-10-26:
Sometimes OTA fails. I have need 2 or 3 firmware update before OTA works fin with some devices.

RavenSystem, 2018-10-27:
@Codi878 Have you tried version 0.4.3?

Codi878, 2018-10-28:
Hi Jose, 

I am out of the house when I'm at home, I will test it.

Best Regards

Codi878, 2018-10-29:
Hi Jose 

Everything works fine on 4.3 firmware. I tested on two Sonoffs.

Thank You

closed 2018-10-29 by RavenSystem