RavenSystem/esp-homekit-devices #64: flashed sonoff basic with success  but cannot find it on homekit

mahdy89, 2018-10-27:
flashed sonoff basic with success  but cannot find it on homekit
tried it with 3 other sonoff
when i connect to wifi.. and choose the update directory and press join..
waiting 10 minutes. and the led didnt flash.. tried it about 7 times with multiple access point
go to home app.. add accessory... the accessory is not found.

RavenSystem, 2018-10-27:
Hello, can you send a screen capture of your initial accessory config?

mahdy89, 2018-10-27:
[image]
[image]
then i put it in the socket.. setup the wifi ( i tried on 3 ap , 2 asus and 1 tp-link)  and wait for 10 minute.. 
nothing found in home app.. at all..and the wifi on the sonif sometime is showing.. but not accessible it say cannot connect when choosing it in the wifi list

RavenSystem, 2018-10-27:
I was asking about this:
https://raw.githubusercontent.com/RavenSystem/ravensystem-media/master/OTA_config.png

mahdy89, 2018-10-28:
> I was asking about this:
> https://raw.githubusercontent.com/RavenSystem/ravensystem-media/master/OTA_config.png
before config
[image]
after config
[image]
join.. wait more than 10 min.. and nothing.. no led flash .. nothing connected to my wifi,..nothing in home app.. tried this on 3 sonoff

RavenSystem, 2018-10-28:
Can you provide logs, please?

mahdy89, 2018-10-28:
> Can you provide logs, please?

how?

RavenSystem, 2018-10-28:
You can use screen command:
`screen /dev/tty.SLAB_USBtoUART 115200`

mahdy89, 2018-10-28:
[bootloop.mp4.zip](https://github.com/RavenSystem/esp-homekit-devices/files/2523082/bootloop.mp4.zip)

yusufd1, 2018-10-28:
I have exactly the same problem. Will try the above and report back thanks.

yusufd1, 2018-10-29:
Hi the problems still exists
after flashing and configuring the wifi access point and OTA fields and hitting join the Sonoff does not show in home app. This is an extract of the serial monitor. please help:

error writing flash
active_cert_sector: 0xf6000
--- ota_get_pubkey
error reading flash
--- ota_get_privkey
error reading flash
--- ota_boot...0
--- ota_write_status
--- ota_load_user_app
ota_repo='RavenSystem/esp-homekit-devices' ota_version='0.0.0' ota_file='sonoff_ravencore.bin'
--- looping
--- ota_set_verify...OFF
--- ota_get_version
--- ota_connect
target IP is 192.30.253.113 bind socket 50079....OK. socket connect to remote....OK. create SSL....OK. set_fd done. SSL to github.com port 443....OK
send request......OK

HomeACcessoryKid/ota@version:"0.1.0"
--- ota_boot...0
--- ota_compare
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/ota/releases/download/0.1.0/certs.sector.sig HTTP/1.1
Host: github.com

--- ota_connect
Fatal exception (9): 
epc1=0x40105cb4
epc2=0x00000000
epc3=0x4020352d
excvaddr=0xa768fce6
depc=0x00000000
excsave1=0x40105c4c
Registers:
a0 40105c4c a1 3fff0270 a2  67697312 a3  3fff89d4
a4  3ffe9724 a5  0000000b a6  00000008 a7  0460c76f
a8  00000003 a9  00010046 a10 0000fff8 a11 00000002
a12 a768fce6 a13 3ffefa5c SAR 00000010

Stack: SP=0x3fff0270
0x3fff0270: 00000103 3ffee0f8 00000046 40105bfe
0x3fff0280: 3fff02c0 00000020 3fff164c 4020581b
0x3fff0290: 00000003 3fff28e4 00000046 4020585e
0x3fff02a0: 402438e5 00000001 00000046 40205888
0x3fff02b0: 002c4288 0fc2ddbc db48f198 40205c7e
0x3fff02c0: 00000008 3fff47c0 3ffeae98 40100ca4
0x3fff02d0: 3ffee0f8 3fff4418 3ffeae98 40246df2
0x3fff02e0: 00000008 3ffee0dc 00000000 00000042

Free Heap: 1734987214
_heap_start 0x3fff18e0 brk 0x3fffd72c supervisor sp 0x40000000 sp-brk 10452 bytes
arena (total_size) 48716 fordblks (free_size) 1734976762 uordblocks (used_size) -1734928046

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   unknown
Flash Mode:   DOUT
Flash Speed:  40 MHz
[... 190 lines cut ...]

>>> wifi_config: Initializing WiFi config
>>> wifi_config: Found configuration, connecting to Muhammad2
user-init-done
mode : sta(bc:dd:c2:0f:98:f1)
add if0
scandone
add 0
aid 1
cnt 

connected with Muhammad2, channel 7
dhcp client start...
ip:192.168.10.103,mask:255.255.255.0,gw:192.168.10.1
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected
wifiready-done
--- ota_boot...0
OTABOOT VERSION: 0.1.0, compiled Apr 15 2018 18:18:30
--- ota_init
error reading flash
error reading flash
active_sector: 0x0
--- ota_set_verify...OFF
--- ota_get_version
--- ota_connect

RavenSystem, 2018-10-29:
In the log now all is ok. No random restart. You must allow to finish the update process.

yusufd1, 2018-10-29:
I fixed it. I just changed the 8MB to 1MB in the esp flash tool script. Maybe it was coincidence but I did wait for many minutes the first time and it did not work. Changing to 1MB did the trick. thanks for the help and the great work you have done on this. please keep up the good work.

closed 2018-11-04 by RavenSystem