RavenSystem/esp-homekit-devices #66: Sonoff 4Ch Pro R2 - only one switch at a time

kevinstiller, 2018-10-28:
Hey there,

first of all, almost every of my sonoff is working pretty well. My dad got a 4Ch for his home, and after I flashed it, we noticed that only 1 switch could be turned on at a time. Is this behaviour intended? 

Best regards and greetings from Germany! Cheers

RavenSystem, 2018-10-28:
Hello @kevinstiller Can you provide more details? What you say for example  is that when switch 1 is turned on, and you turn on switch 2, switch 1 turns off automatically?

kevinstiller, 2018-11-10:
If I turn on 1 switch in homekit, then another one, the first one is turned off again :-(

RavenSystem, 2018-11-10:
No, that is not intended. You can turn on/off each switch independently. You can see that in this video, for example: https://www.youtube.com/watch?v=RCIfNauK8MI

Maybe your DYP switch setup is incorrect. Try this setup:
S6: 1
K5: all 1
K6: all 0

closed 2018-11-26 by RavenSystem