RavenSystem/esp-homekit-devices #67: sonoff 4ch

sercitecotf, 2018-10-29:
Buenas Tardes 
me presento, soy Juan Ramón
y soy nuevo en esto de sonoff :-) he actualizado dos sonoff basic con ravencore 0.43 sin problemas, pero con el sonoff 4ch no se como entrar al menú de configuración para cambiar del basic al 4ch. ¿me pueden ayudar?

Device Type [Default: 1]
Reboot required when changed.

You can choose between those device types:

Switch Basic
Switch Dual
Socket + Button
Switch 4ch
Thermostat
Switch Basic + TH Sensor
Water Valve
Garage Door
Socket + Button + TH Sensor
ESP01 Switch + Button
Shelly 1 -> Not implemented yet

RavenSystem, 2018-10-29:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Sonoff-RavenCore#user-settings

sercitecotf, 2018-10-30:
Buenos días 
Esto lo he visto pero no entiendo en proceso, he probado y probado pero no soy capaz, se que se me pasa algo pero no sé el qué

sercitecotf, 2018-10-30:
ok, ya lo he conseguido 
gracias

sercitecotf, 2018-10-30:
he notado un pequeño inconveniente, después de un reinicio a los valores de fabrica, vuelve a los valores del sonoff basic, en mi opinión se tendría que mantener el tipo de dispositivo reconfigurado, ya que si se quiere cambiar se puede volver hacer desde el setup

RavenSystem, 2018-10-31:
Factory defaults are factory defaults. 

If you want to reset only HomeKit settings, remove your accessory from iOS Home App, and if you want to change your wifi settings, reboot your device and wait 3 minutes until soft AP starts.

closed 2018-11-05 by RavenSystem