RavenSystem/esp-homekit-devices #68: change the GPIO output for normal ESP01 Relaysmodul

heizie, 2018-10-29:
Hi, I've already tried the sonoff_ravencore and otaboot.bin for my ESP01 Relaysmodul. None of them can change the state of GPIO 0 or 2 (i can see the response from homekit but relay didn't works). Any chances to give a update or give me some advise to change the C code?  thx

RavenSystem, 2018-10-31:
In ESP01 (Device type 10) GPIO 0 is used only as button, to allow stateless switch button control and factory reset. Relay must be connected to GPIO 2.

heizie, 2018-11-01:
But it didn't has a button. it looks like this
[image]

I've search it's manual:
[image]

RavenSystem, 2018-11-01:
Sorry, but in this firmware, GPIO0 is used to connect a button, and GPIO2 is used to connect a relay. If you want to use it in other way, you must build your custom firmware adapted to your needs.

closed 2018-11-04 by RavenSystem