RavenSystem/esp-homekit-devices #69: Sonoff

Kristian8606, 2018-10-30:
I have a working sonoff switch that has been reset for no reason and I have to add it to my homekit . I do not know what could be the reason to reset. did anyone have a similar case. Do you think that a disturbance could affect the reset button. I’m sorry my bad english

RavenSystem, 2018-10-31:
Do you add your accessory to HomeKit only? Or you must setup Wifi first?

What firmware version has your Sonoff?

Kristian8606, 2018-11-01:
I add WIFI then add it to my home
The version is Sonoff RavenCore Version 0.4.3

Kristian8606, 2018-11-01:
thank you will report

RavenSystem, 2018-11-04:
Try version 0.4.4. I have released a few minutes ago.

yura2507, 2018-11-27:
Hello! Which firmware to update Sonoff Touch EU 1ch.? 0.4.10 and 0.5.0 do not work. Thanks.

RavenSystem, 2018-11-27:
@yura2507 It should work with device type 1. Can you provide logs indicating when you press button, please?

yura2507, 2018-11-27:
> 
> 
> @yura2507 It should work with device type 1. Can you provide logs indicating when you press button, please?

After the firmware, I do not see the wi-fi device. The firmware ESPEASY_MEGA is recorded and running. (Chip is ESP8285)
esptool.py --port COM3 erase_flash
esptool.py v2.5.1
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
MAC: 60:01:94:84:62:4d
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 5.1s
Hard resetting via RTS pin...

I write down the firmware ESP8266Flasher:
Note:Serial port connected.
Note:Begin find ESP8266.
Note:ESP8266 ACK success.
Note:ESP8266 ACK success.
Note:Set base address success.
Note:Program flash success.
Note:Program flash success.

yura2507, 2018-11-27:
> 
> 
> @yura2507 It should work with device type 1. Can you provide logs indicating when you press button, please?

It was possible to download the firmware by another program. The switch works, the network (ravencore-84623D) has appeared. I could not connect yet :).

[image]

yura2507, 2018-11-27:
> 
> 
> @yura2507 It should work with device type 1. Can you provide logs indicating when you press button, please?

Everything worked out! Thank you!

yura2507, 2018-11-27:
How can I connect a temperature sensor to the device (Sonoff Touch EU 1ch (ESP8285))? What pin?

yura2507, 2018-11-27:
I apologize for the trouble. Everything worked out. GPIO14 - worked.

RavenSystem, 2018-11-27:
@yura2507 

> Could you please advise as for device Sonoff Basic. Is it possible to realize mode of 2 channels on it? One channel we have (is available) and the second like on GPIO14.
That is a very custom setup. A better way is to use a Sonoff Dual with device type 2.

> One more question about frequency (how often) scanning of temperature sensors?
30 seconds.

> Is it possible to realize the function of permanent scanning or at least with 3-10 sec. delay?
Permanent scanning no, it's impossible. But it's possible to add a function to customize time between temperature readings, selecting from 3 to 60 secs, for example. If you are interested, please, create a new issue with that matter.

closed 2018-12-11 by RavenSystem