RavenSystem/esp-homekit-devices #70: Support for Sonoff B1

dbambus, 2018-11-04:
Hey, 

first of all great work :-). RavenCore is working perfectly on many devices in my Home-setup. 

I got a few Sonoff B1 laying around flashed with tasiota here, but i wanna use them wir HomeKit directly, will there any chance of support by RavenCore for them ?

RavenSystem, 2018-11-04:
Sure, but remember that can past even several months until B1 is supported. I will add this to TODO list.

I don’t have a B1, so if you can provide info about gpios and features will be helpful.

RavenSystem, 2018-11-04:
Added to TODO list: https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List

morbidpete, 2018-11-14:
If you make an amazon wish list with the B1 on it I'll buy it for you tomorrow to help get it officially supported. I've been looking at the B1's to use where the basic in the 3d printed wall switch isn't possible.

arturodiazad1, 2018-11-16:
Hello it would be interesting to be able to connect a led strip ws2812b to Sonoff, if it is possible to add it to future projects, I appreciate it

dbambus, 2019-01-30:
Hey, 

Sorry it took me some time to answer. Further Information about the Sonoff B1 is given here http://tinkerman.cat/sonoff-b1-lights-and-shades/

The TX,RX, Vcc, GND and GPIO0 are available to programming without any advanced soldering skills. On the board itself you'll find a ESP8285, basically a 8266 with build in SPI Flash, next to the controller MY9231 for the LEDs itself.

A Library for the My92xx Controller is available at Github https://github.com/xoseperez/my92xx

The development process of the Support of the B1 in the Tasmota firmware is found in this Issue https://github.com/arendst/Sonoff-Tasmota/issues/676 all the Information regarding wiring is given there. 

My problem is I'm not a C-Programmer, just Arduino, shame on me, but for lightbulbs it would be beneficial to have a standalone FW with support of HomeKit. If you need any help, I got 6 Sonoff B1 here.

RavenSystem, 2019-02-28:
Actually not enough time to develop .

closed 2019-02-28 by RavenSystem