RavenSystem/esp-homekit-devices #72: how to get ravencore work with dht11

edmundhei, 2018-11-05:
I manage to install ravencore on esp8266 and add it to homekit. But homekit detect it as a switch. How do I set it to work with dht 11?

hmbarbosa, 2018-11-05:
Follow the instructions and use Eve for HomeKit by Eve Systems GmbH from app store to change. :) cheers

edmundhei, 2018-11-05:
i have look at the wiki and not quite sure how to edit it with the eve app. I don’t see there is any option to change the settings.

mmiglier, 2018-11-05:
“Show Setup” virtual type is not visible by default on the Eve app. There’s an option to make hidden types visible.

edmundhei, 2018-11-05:
How to change that option? where can i change it from?

hmbarbosa, 2018-11-05:
Open Eve, selects Rooms, open your switch and it should be there ”show Setup” move to ”on” and wait the reboot... Refresh Eve.. :)

edmundhei, 2018-11-05:
Ok Thanks for the quick reply. i managed to change the device type to thermostat but i can’t change the sensor type to 1 which is dht -11 . default is22. Also which gpio pin should i connect the dht11 to? Thanks

RavenSystem, 2018-11-05:
Sensor must be connected to GPIO14, VCC and GND. This is the default GPIO used by Sonoff TH and S22.

closed 2018-11-26 by RavenSystem