RavenSystem/esp-homekit-devices #73: Use ESP12E to form sonoff 4ch

yukonx, 2018-11-05:
HI !

Use ESP12E to form sonoff 4ch but press the second button
Will crash, is there a way to rule out?

Sorry, my English is not good.

[https://youtu.be/44CrEA9MwH0](url)

RavenSystem, 2018-11-05:
I don't know where is the problem. All buttons are managed in the same way. Button 2 must be connected to GPIO9, but I understand that you have done it. Can you send longs, please?

yukonx, 2018-11-05:
Can you teach me how to get a log?

RavenSystem, 2018-11-05:
You can use screen command after connecting your device to computer:
`screen /dev/<your_device> 115200`

yukonx, 2018-11-06:
### ### SONOFF 4CH . Working normally but PB2.3.4 no signal response

[https://youtu.be/RCIfNauK8MI](https://youtu.be/RCIfNauK8MI)

### ### ESP12E    PB2 PB2 Will crash

[https://youtu.be/pYG0lSRfCOI](https://youtu.be/pYG0lSRfCOI)

onnected with sonoff, channel 1
dhcp client start...
ip:192.168.0.12,mask:255.255.255.0,gw:192.168.0.1
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected
>>> Creating accessory for type 4
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: C4:E7:BF:EE:CE:51
>>> HomeKit: Found admin pairing with 1C6440B0-4037-43C0-8976-E47730DDE3CA, disabling pair setup
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=RavenCore 445A19 md=RavenCorepv=1.0id=C4:E7:BF:EE:CE:51c#=259s#=1ff=0sf=0ci=8 Port=5556 TTL=4500
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with 1C6440B0-4037-43C0-8976-E47730DDE3CA
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> Toggle Switch 1
>>> Relay 1 -> 1
>>> HomeKit: [Client 4] Update Characteristics
>>> Toggle Switch 2
>>> Relay 2 -> 1
>>> HomeKit: [Client 4] Update Characteristics
>>> Toggle Switch 3
>>> Relay 3 -> 1
>>> HomeKit: [Client 4] Update Characteristics
>>> Toggle Switch 4
>>> Relay 4 -> 1
>>> HomeKit: [Client 4] Update Characteristics
>>> Toggle Switch 1
>>> Relay 1 -> 0
>>> HomeKit: [Client 4] Update Characteristics
>>> Toggle Switch 2
>>> Relay 2 -> 0
>>> HomeKit: [Client 4] Update Characteristics
>>> Toggle Switch 3
>>> Relay 3 -> 0
>>> HomeKit: [Client 4] Update Characteristics
>>> Toggle Switch 4
>>> Relay 4 -> 0
>>> Toggle Switch manual
>>> Relay 1 -> 1

RavenSystem, 2018-11-06:
Sonoff 4ch works normally. But there is a problem with logs, and buttons 2, 3 and 4 don't show anything in logs. This will be fixed in next version.

However, ESP12E crash when you use button attached to GPIO9, and I don't know the reason. Maybe a problem with wires or connection tracks?

RavenSystem, 2018-11-07:
Try version 0.4.7. Now all logs should work right.