RavenSystem/esp-homekit-devices #74: 0.4.4

sercitecotf, 2018-11-06:
He actualizado varios sonoff basic con la 0.4.3 y una vez ha terminado el proceso reviso y tienen la 0.4.4, y después de activar el GPIO14 empieza hacer cosas raras, al pulsar el interruptor hace doble encendido y apagadlo, y también retardo en encendido o apagado.
como puedo poner la 0.4.3 sin que se actualice a la 0.4.4?

sercitecotf, 2018-11-06:
he actualizado uno que tenia con la versión 0.4.3 y ahora hace lo mismo que los otros :-(

RavenSystem, 2018-11-06:
En un rato voy a publicar la versión 0.4.5 con un intento de corregir el problema. Es que estoy usando otro sistema para evitar las interferencias, pero aún hay que hacer más pruebas.

sercitecotf, 2018-11-06:
ok
gracias

RavenSystem, 2018-11-06:
v.0.4.5 publicada. Por favor, actualiza y comenta los resultados. ¡Muchas gracias!

sercitecotf, 2018-11-06:
sigue igual, pulsas el interruptor GPIO14 se enciende si se apaga y aveces no hace nada, si esta encendido y pulsas para apagar no hace nada

sercitecotf, 2018-11-06:
y hace interferencia con otro, al encender y apagar varias veces se enciende los dos o se apagan los dos

sercitecotf, 2018-11-07:
podre actualizar la v 0.4.3 sin que en el proceso se actualice a la 0.4.5?

RavenSystem, 2018-11-07:
No, tienes que esperar un poco a la 0.4.6.

sercitecotf, 2018-11-07:
ok
gracias

sercitecotf, 2018-11-07:
probada y me va bien, por mi parte no le he notado nada raro :-)

closed 2018-11-07 by RavenSystem