RavenSystem/esp-homekit-devices #78: Sonoffs not working remotely

yusufd1, 2018-11-09:
I have 2 Sonoffs on the latest ravencore firmware. They work fine when my iPhone is on the same Wifi network but if I switch TJ LTE then the home app and eve show them as not responding. My Apple TV is connected as a hub and my other sonoffs on hap-node work fine on LTE. This problem has only started on the latest 0.46 firmware. Is this a bug? What can be the problem. Please let me know.

danieljcooper, 2018-11-09:
Since updating to 0.4.6 I have also had issues controlling my devices via HomePod. Siri times out saying it can’t get in touch with the devices. No issues using the Home APP

RavenSystem, 2018-11-09:
Nothing has change with HomeKit management. Try reseting wifi router and devices (Sonoff, AppleTV and HomePod), and remember to set "show setup" to "off" before.

yusufd1, 2018-11-09:
Ok let me try that and revert. Once again keep up all the great work. Can’t wait for the new features.

closed 2018-11-09 by yusufd1