RavenSystem/esp-homekit-devices #79: Request

morbidpete, 2018-11-10:
Sonoff Basic. Have the LED on when the switch is off and LED off when Switch is on. Give the switch a nice glow at night and easier to find.

morbidpete, 2018-11-10:
I'm sorry. Looks like this has been discussed[here](https://github.com/RavenSystem/esp-homekit-devices/issues/71)

So very similar. Just the opposite of what was asked

RavenSystem, 2018-11-10:
If you want to locate a HomeKit device, you can use the built-in identify function in EVE App (clicking on ID in accessory gears at top right corner). I understand your needs, but I can not add many extra functions because HomeKit runs with a lot of resources in a chip (esp8266) that hasn't enough memory to manage much more. However, Tasmota has many custom features, but lacks HomeKit.

Here you can read about:
https://github.com/arendst/Sonoff-Tasmota/issues/3876
https://github.com/arendst/Sonoff-Tasmota/issues/249

morbidpete, 2018-11-10:
Appreciate the the response. Not a big need. A want vs a need. Still amazing work. Thank you!

closed 2018-11-10 by RavenSystem