RavenSystem/esp-homekit-devices #83: Please add support for ds18b20 sensor

faronov, 2018-11-14:
this will be very useful with combined as Water Heater characteristics 
this sensor has models with  a waterproof case

faronov, 2018-11-14:
I have sonoff basic devices and this sensor and can test it if you need some feedback.

yusufd1, 2018-11-15:
Thanks RavenSystem I know you can do it. I have faith in your expertise. This model of sensor is also great for a swimming pool and the user can therefore see very quickly on the home app what the temperature is. We entering summer here in South Africa and this will be great. I had it working on Tasmota so theoretically it should be pretty easy.

rechin304, 2018-11-17:
I have TH16 and ds18b20 ready for testing 👍

ur5zeb, 2018-11-24:
Hi. If you want to add the sensor DS18B20
1. Change the function "temperature_sensor_worker" in the file "main.c"
2. Add lines to file "Makefile"
  extras/ds18b20 \
  extras/onewire \
 3. Select Device Type =6 Select sensor Type 3
  
`void temperature_sensor_worker() {
    float humidity_value, temperature_value;
    ds18b20_addr_t addrs[1];
  //  float temps[1];
    dht_sensor_type_t current_sensor_type;
    int sensor_count;
    if (dht_sensor_type.value.int_value == 1) {
        current_sensor_type = DHT_TYPE_DHT11;
    } else {
        current_sensor_type = DHT_TYPE_DHT22;
    }
    sensor_count = ds18b20_scan_devices(SENSOR_GPIO, addrs, 1);
    printf(">>> Sensor: DS18B20 %g\n", sensor_count);
    if (dht_read_float_data(current_sensor_type, SENSOR_GPIO, &humidity_value, &temperature_value)  ||  (sensor_count > 0)) {
       if (dht_sensor_type.value.int_value != 3){
        printf(">>> Sensor: temperature %g, humidity %g\n", temperature_value, humidity_value);
        
        if (temperature_value != old_temperature_value) {
            old_temperature_value = temperature_value;
            current_temperature.value = HOMEKIT_FLOAT(temperature_value);
            homekit_characteristic_notify(&current_temperature, current_temperature.value);
            
            if (humidity_value != old_humidity_value) {
                old_humidity_value = humidity_value;
                current_humidity.value = HOMEKIT_FLOAT(humidity_value);
                homekit_characteristic_notify(&current_humidity, current_humidity.value);
            }
            
            if (device_type_static == 5) {
                update_state();
            }
        }
       }else{
           temperature_value = ds18b20_read_single(SENSOR_GPIO);
           printf(">>> Sensor: temperature %g \n", temperature_value);
           if (temperature_value != old_temperature_value) {
               old_temperature_value = temperature_value;
               current_temperature.value = HOMEKIT_FLOAT(temperature_value);
               homekit_characteristic_notify(&current_temperature, current_temperature.value);
               }
       }
        
    }else {
        printf(">>> Sensor: ERROR\n");
        led_code(LED_GPIO, SENSOR_ERROR);
        
        if (current_state.value.int_value != 0 && device_type_static == 5) {
            current_state.value = HOMEKIT_UINT8(0);
            homekit_characteristic_notify(&current_state, current_state.value);
            
            relay_write(false, relay1_gpio);
        }
    }
}
`

ur5zeb, 2018-11-24:
@RavenSystem  I do not use a thermostat, I just had to measure the temperature.
I showed how to connect the sensor
When is planning to release a new firmware with the support of "DS18B20"?

RavenSystem, 2018-11-24:
Very soon. Code is ready, and feature will be launched with version 0.5.0, and if there is not any new bug, I hope it will be released in a couple of days.

RavenSystem, 2018-11-26:
v0.5.0 released just now.

rechin304, 2018-11-27:
works ok,  but future improvement that i suggest is if sensor type 3 is selected not to create humidity device and readings.

RavenSystem, 2018-11-27:
@rechin304 Humidity service will can selectable to remove with an additional option in a future version.

38 C limit: HomeKit framework manages temps from -100 to 200 C, and sensor is implemented with the eep-open-rtos driver, and I think that there is not limited to 38 C. remember that temperature is reading every 30 seconds. Can you check it again knowing this info, please?

rechin304, 2018-11-27:
Is not about readings (readings are ok) is about when used a Thermostat you cannot increase the temp more than 38 C when in heat mode. Can be a homekit limitation i imagine as thermostats are most of the time used for heating homes and not boiler water.

closed 2018-11-30 by RavenSystem