RavenSystem/esp-homekit-devices #86: Sonoff does not show temperatures below 0 Celsius

Codi878, 2018-11-17:
Hi Jose,

I have a problem with the device type 6. Sonoff TH is placed outside and measures temperature and humidity and controls the lighting outside. Everything worked great until the temperature dropped below 0 degrees Celsius. 
Attached are some photos of how it looks in the application. Firmware version 3.4

Best Regards

[image]
[image]
[image]

RavenSystem, 2018-11-17:
Yes, it's annoying. Official Apple HomeKit Protocol set Temperature between 0 and 100, as you can see in this document: 
https://developer.apple.com/homekit/specification/

Here is a thread about this matter: https://github.com/KhaosT/HAP-NodeJS/issues/147

I'm going to set minimum temperature to -100, ignoring Apple specifications.

RavenSystem, 2018-11-17:
Update to version 0.4.9, please.

Codi878, 2018-11-18:
Hi Jose

On 4.9 firmware everything works fine :) Thank You for your quick fix.

[image]

RavenSystem, 2018-11-18:
Thanks to you for your feedback.

closed 2018-11-18 by RavenSystem