RavenSystem/esp-homekit-devices #89: T1-2C

egorist, 2018-11-19:
Sorry, but the device begins to on/off randomly again. Only main button. With advanced button everything is good.

RavenSystem, 2018-11-27:
Try version 0.5.1.

egorist, 2018-11-28:
I see the is new repository. I need to setup new repository in ota firmware again? Because I can’t set new firmware by ota now.

RavenSystem, 2018-11-28:
It’s automatic. If you are running a version prior to 0.4.10, you must update twice to get version 0.5.1 and above:
Old_version -> 0.4.10 -> 0.5.1

v0.4.10 changes ota repo automatically.

egorist, 2018-11-28:
Yes, I done it, thanks. Need some days for testing

egorist, 2018-11-29:
2 times device switches on randomly for this time. So update doesn’t work ((

RavenSystem, 2018-11-30:
Try version 0.5.2.

egorist, 2018-12-05:
For now, It works, thanks)

closed 2018-12-05 by RavenSystem